SlimeVRX/SlimeVRX #9: [HW] IMU-Comparison

NoLoPhe, 2022-04-06:
IMU-Comparison [Follow SlimeVR](https://docs.slimevr.dev/diy/imu-comparison.html)

Thuần nghiên cứu hay Project

BNO080/BNO085: CN¥280 [BUY](https://item.taobao.com/item.htm?id=16838528836)
BNO055: 22.25 USD ~ CN¥141.54 [BUY](https://www.arrow.com/en/products/a000070/arduino-corporation)
ICM-20948:  CN¥67 [BUY](https://item.taobao.com/item.htm?id=570953772841)

NoLoPhe, 2022-04-06:
Nên chọn cái tốt nhất (đắt nhất), hay là tối ưu những cái giá thấp hơn
BNO080 được tối ưu bởi CEVA tốt nhất hiện nay, khả năng chống trôi đã được kiểm nghiệm bới SlimeVR, thời gian reset 30-60p.
Giá nó quá cao: 1tr VND / 1 pcs.
Xin hỗ trợ từ HiPNUC 6 cảm biến, điều kiện kinh phí thiếu thốn, xin hỗ trợ từ nhiều nguồn.
Hướng: nghiên cứu thuần gồm 2 mảng lớn là IMU và mạng neural Deeplearning, muốn có một bài báo, theo hướng nghiên cứu sau đại học, cải thiện tiếng Anh, theo hướng nghiên cứu độc lập, tự mở lab, không muốn bị bóc lột sức lao động, muốn nghiên cứu chuyên sâu một mảng, kiến thức nền vững.

NoLoPhe, 2022-04-12:
Em chào thầy ạ!

Em muốn hỏi đề tài chỉ sử dụng cảm biến IMU, không có cơ cấu chấp hành (vd: cánh tay robot) thì có được chấp nhận không, vì đề tài này là tái tạo lại kết quả một paper, rất nhiều vấn đề phải tìm hiểu, nếu thêm phần cơ cấu chấp hành thì lượng công việc phải làm sẽ tăng lên nhiều.

Với đề tài: "Ước tính cơ thể người từ 6 cảm biến quán tính" em băn khoăn đi theo hướng thuần nghiên cứu hay Project.
Em muốn: Viết nhiều tài liệu, viết một bài báo, chuẩn bị cho sau đại học, em rất muốn đi theo hướng thuần nghiên cứu.

Em đã đi làm 1,5 năm ở một công ty về xử lý ảnh, detect người, xe, mũ bảo hiểm, khẩu trang, segmentation đám cháy. Công ty cũng tạo điều kiện để em làm dự án cá nhân ước tính cơ thể người từ camera chiều sâu. Làm ở công ty dự án thay đổi rất nhanh, code rất nhiều, học được nhiều. Em muốn đi chậm lại, học và viết tài liệu nhiều hơn, từng bước chắc chắn ạ.

Em muốn nghiêm túc thực hiện đồ án tốt nghiệp. Em tạm chia đề tài thành 2 phần:
- Đầu vào cảm biến IMUs
- Mạng deep neural network

Về cảm biến IMU, em đã học từ zero từ chuỗi bài của cty HiPNUC trên Zhihu và translate nó qua tiếng Việt, em rất thích chuỗi bài này vì nó có rất nhiều code để thực hành. Sau mỗi phần lý thuyết có phần code, nếu phần lý thuyết quá nhiều công thức thì em sẽ đi vào code, sau đó sẽ đọc lại lý thuyết từ từ. Việc translate tài liệu sang tiếng Việt cũng giúp em hiểu tài liệu sâu hơn, giống như viết ra giấy một lần vậy ạ.
Có một tin vui là em đã xin hỗ trợ từ cty HiPNUC 6xHi229 9-axis (dự kiến 20/4 sẽ ship), em đang translate datasheet Hi229 từ tiếng Trung sang tiếng Anh, thầy xem qua và add issues giúp em với ạ[Link]. Cùng với việc translate thì em cũng tìm hiểu sâu hơn về các thông số của cảm biến để translate cho đúng, có rất nhiều thông số rất mới với em.

Về phần mạng neural neural network, trước tiên em đã chạy được code, mô phỏng kết quả trực tiếp bằng thư viện Open3d ở python[1] hoặc gửi dữ liệu sang Unity[2]. Cũng có rất nhiều vấn đề phải tìm hiểu ở phần này, em mong muốn tự đào tạo lại mạng, thay vì 6 cảm biến hiệu tại thì tăng lên 8 cảm biến (thêm 2 cảm biến ở bàn chân).
Em có thể mô tả việc traning lại model như sau:
- Dữ liệu đầu vào model không phải là dữ liệu real IMU.
- Dữ liệu đầu vào: Acceleration and Rotatio được suy ra từ công thức từ bộ dữ liệu action: AMASS dataset.
- Hình
- Mỗi cột là một chuỗi chuyển động liên tiếp
- Input data: https://github.com/SlimeVRX/SlimeVRX/tree/main/transpose
- Acceleration and Rotatio được tính từ 6 vị trí tương ứng trên cơ thể người: left forearm, right forearm, left lower leg, right lower leg, head, pelvis(root).

Công việc của chúng ta là từ dữ liệu real IMU, chuyển đổi thành Acceleration (đã có) and Rotatio (tính theo hướng dẫn ở phụ lục cuối paper). Tìm hiểu mô hình người SMPL(được dùng trong AMASS dataset). Về cảm biến IMU có vấn đề về drift (lỗi tích lũy của cảm biến sau một thời gian, vị trí là tích phân vận tốc góc, vận tốc góc là tích phân của gia tốc) cũng phải được giải pháp hạn chế.

Thầy ơi, bọn em không có nhiều kinh phí nên có thể xin thêm tài trợ ở đâu ạ.

Theo khảo sát SlimeVR (IMU-Comparison):
- BNO080/BNO085 (best): CN¥280 ~ 1.005.493 VND
- ICM-20948: CN¥67 ~ 240.600 VND
- BNO055: 22.25 USD ~ CN¥141.54  ~ 508.277 VND
ngoài ra Hi229 xin hỗ trợ từ HiPNUC:
- Hi229: CN¥199 ~ 714.618 VND

Phần cứng hiện tại đã có:
- 3 cảm biến BMX055 9-dof
- 1 server training 4xGPU Tesla M40 24GB (em mua lúc làm việc ở cty, nghỉ việc nên em để bụi được 1 năm rồi, server rất ồn)

Em muốn nhắm mắt lại, em sẽ hình dung được những gì mình đang làm và biết tiếp theo mình làm gì.

NoLoPhe, 2022-04-14:
Hướng thuần nghiên cứu: em muốn nghiên cứu trong doanh nghiệp (làm R&D trong Industry) hay theo hướng học thuật (Academia)? Nếu theo hướng học thuật thì thầy có thể giới thiệu em với network của các thầy trong bộ môn để em có thể học tiếp lên Master và PhD.

chưa định hướng được công việc
tính bốc đồng cao
không giỏi giao tiếp
làm project công việc thay đổi quá nhanh. 
Học, viết bài, tài liệu đầy đủ, từng bước vững vàng.

việc phải làm trong ngày: 
- trả lời câu hỏi của thầy
- so sánh các cảm biến
- mắc xích giữa dữ liệu thật chuyển đổi thành đầu vào model
- thử training một model pytorch.

NoLoPhe, 2022-04-14:
Công việc lỡ dỡ:
- Đi sửa server GPU
- Học bằng Toeic
- Học bằng Tin học

NoLoPhe, 2022-04-14:
Hiệu chỉnh cảm biến
There are two types of calibration: static, i.e. correction of parameters not variable over time such as gain, and dynamic,

NoLoPhe, 2022-04-18:
Em chào thầy ạ!

Em là Nguyễn Thành Tín lớp 15CDT2 khoa Cơ Khí (không phải khoa Công Nghệ Thông Tin)
Em vừa nhận được thông báo được nhận đồ án tốt nghiệp cách đây 2 tuần (thời gian 04/04/22-18/07/22).
Em muốn nhờ thầy hỗ trợ hướng dẫn Đồ án tốt nghiệp giúp nhóm chúng em.

Đề tài em chọn là ước tính cơ thể người từ 6 cảm biến quán tính.
Em đang tái tạo lại kết quả của bài báo "Transpose: Real-time 3D Human Translation and Pose Estimation with Six Inertial Sensors". (Link1)
Bài báo sử dụng IMU chuyên dụng hãng Noitom để thử nghiệm, em đang tìm giải pháp phần cứng thay thế.
Thầy hướng dẫn bọn em: Thầy Trần Quang Khải (khoa Cơ Khí) đã cho phép bọn em thực hiện đề tài này.

Em tìm được 2 bài báo của thầy:
- "Human action recognition method based on conformal geometric algebra and recurrent neural network."(2020)
- “LSTM for Human Activity Recognition Based on Feature Extraction Method Using Conformal Geometric Algebra”. In proceeding of FAIR 2021.

Em nhận thấy có rất nhiều điểm tương đồng trong những nghiên cứu của thầy, em cần sự giúp đỡ của thầy.

Em xin giới thiệu thêm về bản thân và công việc đã làm. Em đã đi làm 1,5 năm ở một công ty (03/2019-07/2020) về xử lý ảnh, detect người, xe, mũ bảo hiểm, khẩu trang, segmentation đám cháy. Công ty cũng tạo điều kiện để em làm dự án cá nhân ước tính cơ thể người từ camera chiều sâu. Làm ở công ty dự án thay đổi rất nhanh, code rất nhiều, học được nhiều. Nhưng em muốn đi chậm lại, học và viết tài liệu nhiều hơn, từng bước chắc chắn. Đây là CV của em trước khi vào công ty [Link]

Em muốn nghiêm túc thực hiện đồ án tốt nghiệp. Đề tài có 3 phần quan trọng:
1. Đầu vào model là dữ liệu 6 cảm biến IMUs, IMU đang gặp vấn đề về sự trôi (drift). Hướng giải quyết hạn chế sự trôi bằng cách tăng tần số đọc IMU (>100Hz) bằng phần cứng FPGA và sử dụng Kalman filter. Dữ liệu đầu vào có cấu trúc như sau: (Link2)
2. Tạo dữ liệu training Model: Tìm hiểu về bộ dự liệu AMASS, các thông số của mô hình người 3D SMPL.
3. Model sử dụng mạng LSTM viết bằng Pytorch, chuyển sang ONNX Runtime để tối ưu tốc độ cho CPU (tốt nhất nên có GPU hỗ trợ).

Các kết quả thử nghiệm model:
1. Theo kết quả paper của nhóm tác giả, model chạy real-time 90 FPS trên Laptop Intel(R) Core(TM) CPU i7-10750H và NVIDIA RTX2080 Super.
2. Kết quả thử nghiệm trên Laptop của em real-time 10 FPS chỉ với CPU i5-5200U không có GPU.
3. Kết quả được thể hiện trong Unity3D ở tần số 60Hz (Link3)
4. Em đã viết một hướng dẫn để chạy code (Link4)

Hướng áp dụng của đề tài: 
1. Một hệ thống motion capture từ 6 cảm biến quán tính.
2. Áp dụng vào VRChat hỗ trợ Full Body Tracking.

Rất mong được sự hỗ trợ từ thầy. Em xin cám ơn ạ!

Link 1 webpage Transpose paper: https://xinyu-yi.github.io/TransPose/

Link 2 model input data: https://github.com/SlimeVRX/SlimeVRX/tree/main/transpose

Link 3 results displayed in Unity3D at 60Hz: https://twitter.com/Tn23008624/status/1500153059066146816

Link 4 a tutorial to run the code: https://github.com/SlimeVRX/TransPose/blob/main/install_note.md

NoLoPhe, 2022-04-18:
Em chào thầy ạ!

Với đề tài: "Ước tính cơ thể người từ 6 cảm biến quán tính" em băn khoăn đi theo hướng thuần nghiên cứu hay Project.
Em muốn: Viết nhiều tài liệu, viết một bài báo, chuẩn bị cho sau đại học, em rất muốn đi theo hướng thuần nghiên cứu.

Em tạm chia đề tài thành 2 phần:
- Đầu vào cảm biến IMUs
- Mạng deep neural network

Về cảm biến IMU, em đã học từ zero từ chuỗi bài của cty HiPNUC trên Zhihu và translate nó qua tiếng Việt [Link1], em rất thích chuỗi bài này vì nó có rất nhiều code để thực hành. Sau mỗi phần lý thuyết có phần code, nếu phần lý thuyết quá nhiều công thức thì em sẽ đi vào code, sau đó sẽ đọc lại lý thuyết từ từ. Việc translate tài liệu sang tiếng Việt cũng giúp em hiểu tài liệu sâu hơn, giống như viết ra giấy một lần vậy ạ.
Có một tin vui là em đã xin hỗ trợ từ cty HiPNUC 6xHi229 9-axis (dự kiến 20/4 sẽ ship), em đang translate datasheet Hi229 từ tiếng Trung sang tiếng Anh. Cùng với việc translate thì em cũng tìm hiểu sâu hơn về các thông số của cảm biến để translate cho đúng, có rất nhiều thông số rất mới với em.

Về phần mạng Deep Neural Network, trước tiên em đã chạy được code, mô phỏng kết quả trực tiếp bằng thư viện Open3D ở Python[Link3] hoặc gửi dữ liệu sang Unity[Link4]. Cũng có rất nhiều vấn đề phải tìm hiểu ở phần này, em mong muốn tự đào tạo lại mạng, thay vì 6 cảm biến hiệu tại thì tăng lên 8 cảm biến (thêm 2 cảm biến ở bàn chân).
Em có thể mô tả chuẩn bị dữ liệu traning model như sau:
- Dữ liệu đầu vào model không phải là dữ liệu real IMU.
- Dữ liệu đầu vào: Acceleration and Rotatio được suy ra từ công thức từ bộ dữ liệu action: AMASS dataset.
- [image]
- Mỗi cột là một chuỗi chuyển động liên tiếp
- Input data [Link5]
- Acceleration and Rotatio được tính từ 6 vị trí tương ứng trên mô hình cơ thể người SMPL: left forearm, right forearm, left lower leg, right lower leg, head, pelvis(root).

Công việc của chúng ta là từ dữ liệu real IMU, chuyển đổi thành Acceleration (đã có) and Rotatio (tính theo hướng dẫn ở phụ lục cuối paper). Về cảm biến real IMU có vấn đề về drift (lỗi tích lũy của cảm biến sau một thời gian, vị trí là tích phân vận tốc góc, vận tốc góc là tích phân của gia tốc) cũng phải được giải pháp hạn chế.

Đặc biệt vấn đề toán trong mạng Deep neural network em cũng cần thảo luận cùng thầy nhiều hơn. Em đang dịch một cuốn sách về Pytorch (không có bản quyền), nó có quá nhiều phần, em hơi mô hồ và không quá chuyên sâu phần này [Link6]. Em làm việc nhiều hơn với triển khai model (sử dụng pretrain) và chưa training một model (lớn) nào trước đây. Thầy có lời khuyên nào cho em về việc học, viết tài liệu, training một model không ạ. Pooling layer, Fully connected layer, Dense layer, hàm loss, em chưa thể hình dung được. Em muốn nhắm mắt lại, hình dung được những gì mình đang làm và biết tiếp theo mình làm gì.

Thầy ơi, bọn em không có nhiều kinh phí nên có thể xin thêm tài trợ ở đâu ạ.

Theo khảo sát SlimeVR (IMU-Comparison) [Link7]:
- BNO080/BNO085 (best): CN¥280 ~ 1.005.493 VND
- ICM-20948: CN¥67 ~ 240.600 VND
- BNO055: 22.25 USD ~ CN¥141.54  ~ 508.277 VND
- Hi229: CN¥199 ~ 714.618 VND (xin hỗ trợ từ HiPNUC, không có trong danh sách)

Giới thiệu thêm về SlimeVR: Thầy biết VRChat không ạ, đó là một game VR nơi mọi người nói chuyện, đi thăm quan các thế giới 3D. Mặc định VRChat chỉ hỗ trợ tracking 3 điểm là Headset và 2 tay cầm, SlimeVR hỗ trợ fullbody tracking sử dụng cảm biến quán tính. SlimeVR đã huy động từ Crowdsupply được 3,575 backers với 863,825$

Phần cứng hiện tại đã có:
- 3 cảm biến BMX055 9-axis
- 1 server training 4xGPU Tesla M40 24GB (em mua lúc làm việc ở cty, nghỉ việc nên em để bụi được 1 năm rồi, server rất ồn ào)

Link1: chuỗi bài trên Zhihu (https://github.com/SlimeVRX/SlimeVRX/tree/main/hipnuc/document)
Link2: translate datasheet (https://github.com/SlimeVRX/SlimeVRX/blob/main/hipnuc/datasheet/hi229um_cn.md)
Link3: Video Open3D (https://twitter.com/Tn23008624/status/1494618444821839877)
Link4: Video Unity (https://twitter.com/Tn23008624/status/1500153059066146816)
Link5: model input data (https://github.com/SlimeVRX/SlimeVRX/tree/main/transpose)
Link6: Python book: (https://slimevrx.github.io/html/1_Contents.html)
Link7: IMU-Comparison (https://docs.slimevr.dev/diy/imu-comparison.html)

closed 2022-04-20 by SlimeVRX

SlimeVRX, 2022-05-08:
Báo cáo công việc trong 2 tuần (25/04 - 08/05)

Hardware:
- Nhận cảm biến Hi229 (HiPNUC hỗ trợ toàn bộ chi phí, Hi229+Mother Board + vận chuyển)
- Phần cứng hỗ trợ: 6 cảm biến được kết nối vào chung hub 6 cổng USB, truyền dữ liệu về máy tính thông qua dây USB nối dài 5m.

Software:
Đọc dữ liệu từ cảm biến:
- Tạo auto setting tool các lệnh AT - thay đổi nhanh thiết lập dữ liệu cảm biến cần xuất ra.
- Sửa đổi code get data của HiPNUC để lấy đồng thời dữ liệu 6 cảm biến.
- Liên hệ HiPNUC cập nhật firmware hỗ trợ 60 FPS (trước đó hỗ trợ 50-100-200 FPS).
- Do phần cứng hiện đang dùng CPU (không có GPU) load model chỉ đạt 10 FPS không chạy real-time được. Biện pháp: thực hiện ghi dữ liệu ở 60FPS sau đó truyền vào model.

Xử lý dữ liệu Acceleration và Rotation matrix :
- Đọc code live_demo.py và thực hiện các bước: 1. Put imu (x = Left, y = Up, z = Forward) 2.Stand straight in T-pose. Bước đầu không hiểu nhưng thực hiện đúng theo hướng dẫn. Sau rất nhiều thử nghiệm đã có demo đầu tiên.
- Đã gửi issues cho tác giả bài báo Transpose - Xinyu-Yi, thêm wechat để có thể nhận được hỗ trợ nhanh.

Tìm hiểu model SMPL hoạt động:
- Bước đầu tìm hiểu được cách model SMPL hoạt động, tìm hiểu mối liên hệ giữa các hệ tọa: Local -> Global -> Mô phỏng.
- Hệ tọa độ SMPL có hướng (x = Left, y = Up, z = Forward).
- Khi setting các trục Rotation matrix về 0, model SMPL ở tư thế đứng hình chữ T.
- Giả dữ liệu, tạo chuyển động quanh trục Z 90 độ (Rotation matrix) ở khớp vai, nhận thấy cánh tay xoay theo. Khi xoay các trục của Rotation matrix thì khớp xương tương ứng xoay. Vậy có thể tạo các tư thế từ Rotation matrix. 
- Dữ liệu Acceleration chưa được dùng, có thể sẽ liên quan khi cơ thể di chuyển.

SlimeVRX, 2022-06-05:
Dự định ban đầu bọn em sử dụng hệ thống Vive Lighthouse (sóng ánh sáng - laser hồng ngoại) để định vị trong nhà, nhưng đặc điểm của sóng ánh sáng bị chặn bởi vật cản. Bọn em đã tìm hiểu về sóng vô tuyến và tìm thấy hai hệ thống định vị: UWB (Apple Airtag) và Bluetooth AOA.

UWB (Ultra-Wideband) kĩ thuật TOF để xác định vị trí, AOA xác định góc

Nhưng bên bọn em đã liên hệ với một số bên với mong muốn kết hợp với hệ động định vị Indoor vào Motion Capture
- NoopLoop: độ chính xác +- 5 cm, em đã liên hệ, chỉ nói chuyện được với bên bán hàng, chưa liên hệ được với kĩ sư
- Team ULoc: ULoc là một paper em đánh giá là đột phá về UWB, hệ thống của họ có thể theo dõi 3D, mong muốn của họ là áp dụng vào hệ thống motion capture, độ chính xác vật cố định là 3cm và vật chuyển động 8cm, cách bố trí Anchor của họ hình L cũng rất đặc biệt. Họ có opensource một phần nhưng để tái tạo được kết quả của họ rất khó khắn. 
Sau khi gửi mail và kết bạn trên Twitter thì em nói chuyện được với một thành viên trong nhóm thì nhận được vài thông tin, họ đã đăng ký bản quyền công nghệ, mình khó mà phát triển tiếp từ kết quả của nhóm, mình chỉ lấy được ý tưởng và kết quả đánh giá trong bài báo.
- Team VietNam: em rất bất ngờ vì có một team Việt Nam nghiên cứu và ra sản phẩm về UWB, em đã gọi điện trao đổi, họ nói có thể đạt được độ chính xác 5cm nếu sử dụng 9 anchor và hiểu chuẩn kĩ. Họ ưu tiên về độ ổn định trong cộng nghiệp, họ rất khó cho mình đụng vào phần source. Lúc đầu họ không tin mình lắm ạ, họ hỏi về thầy hướng dẫn. Sau đó họ phản hồi rằng có quan tâm đến công nghệ motion capture, và em gửi CV cho họ, nên có khả năng kết hợp. Nếu trường hợp tệ nhất thì em mong muốn họ setup sẵn một hệ thống và em sẽ vào trong đó làm thí nghiện khả năng công nghệ, lấy kết quả ạ. Em muốn tập trung vào nghiên cứu phần core hơn làm sản phẩm, core của mình sử dụng bộ dataset có bản quyền, em không chắc mang lại kết quả như họ mong muốn nên không dám nhận đầu tư. Em đang băng khoăn chỗ này, 
- Cuối cùng là team HiPNUC, lúc đầu em hỏi thì họ nói có dùng qua, em không ngờ search trên Zhihu thì thấy họ có bài viết về UWB đặc biệt kết hợp với IMU để làm mịn, em nghĩ là em ưu tiên nghiên cứu với team HiPNUC, team họ chia sẻ rất nhiều tài liệu, nên trao đổi được với họ nhiều hơn ạ! 

Phần 2: Cắt bỏ
model bao gồm có 5 model nhỏ Pose-S1, Pose-S2, Pose-S3 liên quan đến phần ước tính pose, Trans-B1, Trans-B2 liên quan đến ước tính di chuyển trong 3D, em sẽ bỏ đi phần này trong model, thay vì ước tính, em sẽ dùng trực tiếp vị trí được lấy từ hệ thống định vị UWB.

Em muốn xin thầy hỗ trợ board WB1000, 16.25$/ 1pcs, bọn em cần 5 board để nghiên cứu, học hỏi về công.