SlimeVRX/SlimeVRX #10: [HW] HI229 User Manual

NoLoPhe, 2022-04-07:
[image]

NoLoPhe, 2022-04-07:
[image]
Page: 49 / 58
[Link](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)

NoLoPhe, 2022-04-07:
[image]

Page: 32/117
[Link](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf)

NoLoPhe, 2022-04-07:
[image]

Page 30/58
[Link](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)

NoLoPhe, 2022-04-07:
[image]

Key features of integrated sensors:
Các tính năng chính của cảm biến tích hợp:

Page: 2/117
[Link](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf)

NoLoPhe, 2022-04-07:
[image]

Supply voltage
Page: 45/58

NoLoPhe, 2022-04-07:
[image]

Maximum output rate
Page: 50/58
[Link](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)

NoLoPhe, 2022-04-07:
[image]

NoLoPhe, 2022-04-07:
[image]

NoLoPhe, 2022-04-07:
[image]

NoLoPhe, 2022-04-07:
[image]

NoLoPhe, 2022-04-07:
[image]

NoLoPhe, 2022-04-07:
[image]

SlimeVRX, 2022-04-07:
The Hi229 is a low cost, high performance, small size, low latency Attitude and Heading Reference System (AHRS). This product integrates a 3-axis accelerometer, 3-axis gyroscope, 3-axis magnetic field sensor and a microcontroller

SlimeVRX, 2022-04-07:
HI229是超核电子推出的一款低成本、高性能、小体积、低延时的航姿参考单元（AHRS），本产品集成了三
轴加速度计、三轴陀螺仪、三轴磁场传感器和一款微控制器。

SlimeVRX, 2022-04-07:
The HI229 launched by 超核电子 is a low-cost, high-performance, small size, and low-latency attitude reference unit (AHRS). This product integrates a triaxial accelerometer, a triaxial gyroscope, a triaxial magnetometer and a microcontroller.

SlimeVRX, 2022-04-07:
2 可輸出經過感測器融合算法計算得到的基於當地地理坐標的三維方位數據，包含航向角，俯仰角和橫滾角。

SlimeVRX, 2022-04-07:
It can output 3D orientation data based on local geographic coordinates calculated by the sensor fusion algorithm including Yaw, Pitch and Roll angles.

SlimeVRX, 2022-04-07:
It can output three-dimensional azimuth data based on local geographic coordinates calculated by the sensor fusion algorithm, including heading angle, pitch angle and roll angle.

SlimeVRX, 2022-04-07:
HI229是超核电子推出的一款低成本、高性能、小体积、低延时的航姿参考单元（AHRS），本产品集成了三
轴加速度计、三轴陀螺仪、三轴磁场传感器和一款微控制器。可输出经过传感器融合算法计算得到的基于当
地地理坐标的三维方位数据，包含航向角，俯仰角和横滚角。同时也可以输出校准过的原始的传感器数据。
本产品具有一定的室内地磁抗干扰性能，在一定强度的地磁场干扰环境下仍可正常工作。

SlimeVRX, 2022-04-07:
It can also output calibrated raw sensor data.

NoLoPhe, 2022-04-08:
The Hi229 manufactured by 超核电子 is a System in Package (SiP) that integrates a triaxial 12-bit accelerometer with a range of ±8g, triaxial 16-bit gyroscope with a range of ±2000 degrees per second, a triaxial geomagnetic sensor, and a 32-bit ARM® Cortex™-M4 microcontroller.

NoLoPhe, 2022-04-08:
The firmware provides the sensor fusion algorithm to process sensor data and provide precise real-time 3D orientation, heading, calibrated acceleration and calibrated angular velocity, as well as more advanced contextual outputs.
It can output three-dimensional azimuth data based on local geographic coordinates calculated by sensor fusion algorithm, including heading angle, pitch angle and roll angle. It can also output calibrated raw sensor data.

The firmware provides sensor fusion algorithm to process sensor data and provide precise real-time 3D orientation, heading angle, pitch angle and roll angle, as well as calibrated raw sensor data.

NoLoPhe, 2022-04-08:
The firmware provides sensor fusion algorithm to process sensor data and provide precise real-time 3D orientation, heading, calibrated acceleration and calibrated angular velocity, as well as calibrated raw sensor data.

NoLoPhe, 2022-04-08:
The Hi229 has certain indoor geomagnetic anti-interference properties, and can still work normally under a certain intensity of geomagnetic interference environment.

The Hi229 has certain indoor magnetic anti-interference properties, and can still work normally under a certain intensity of magnetic field interference environment.

NoLoPhe, 2022-04-08:
典型应用:
VR\动作捕捉
高动态环境下姿态测量\运动性能评估
无人机控制

NoLoPhe, 2022-04-08:
Typical application:
- VR\Motion Capture
- Attitude measurement\motion performance evaluation in high dynamic environment
- Drone control

NoLoPhe, 2022-04-08:
[image]

Page: 28/117
[Link](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf)

NoLoPhe, 2022-04-08:
Functional Description

Sensor Configuration

Sensor | Range
--- | ---
Gyroscope | ± 2000°/s
Accelerometer | ± 8G
Magnetometer | 800mG  (miligauss)

NoLoPhe, 2022-04-08:
[image]

Page: 10/58
[Link](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)

NoLoPhe, 2022-04-08:
## Hi229 Connectivity

The Hi229 can support connections to a host microcontroller through various serial interfaces:
- UART interface (TTL 1.8V-5.0V)
- USB (with USB evaluation board)

### Pin Descriptions

Pin Number | Name | Description
--- | --- | --- 
5 | N/C | Reserve
6 | VDD | Supply voltage (sensors) 3.3V
7 | SYNC_OUT | The data output is synchronized with internal pull-up. High level when there is no data output, low level when a frame of data starts to be sent, and return to high level after a frame of data is sent. Need to be suspended when not in use.
8 | RXD | The module serial port receives UART RXD (TXD connected to MCU)
9 | TXD | The module serial port sends UART TXD (RXD connected to MCU)
10 | SYNC_IN | The data input is synchronized with internal pull-up. When the module detects a falling edge, it will output a frame of data. Need to be suspended when not in use.
11 | N/C | Reserve
19 | GND | GND
20 | RST | Reset, internal pull-up. >10uS low level reset module. No need for external resistors and capacitors. It is recommended to connect to the GPIO pin of the MCU for software reset.
21 | N/C | Reserved
22 | N/C | Reserved
23 | N/C | Reserved
24 | GND | GND
25 | N/C | Reserved

NoLoPhe, 2022-04-08:
## Sensor Data Processing

### Attitude angle output accuracy

Attitude angle | Typical value
--- | ---
Roll Angle\Pitch Angle - Static Error | 0.8°
Roll Angle\Pitch Angle - Dynamic Error | 2.5°
Heading angle accuracy in motion (6-axis mode, 30min, horizontal smooth motion, sweeper-like motion mode) | <10°
Heading angle accuracy in motion (In 9-axis mode, no magnetic interference, after calibration) | 3°

NoLoPhe, 2022-04-08:
[image]

Page: 49/58
[Link](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)

NoLoPhe, 2022-04-09:
## Calibration and Interpretation

### Accelerometer and Gyroscope Calibration

The accelerometer and gyroscope are calibrated for scale factor error, non-quadrature error and zero bias error before leaving the factory. Calibration parameters are stored inside the module.
The gyroscope is also calibrated for temperature compensation before leaving the factory. Calibration parameters are stored inside the module.

### Magnetometer Calibration

The magnetic sensor (supported by some models) is ellipsoid calibrated before leaving the factory, but the magnetic sensor is easily disturbed by the magnetic field of the external environment, and generally requires the customer to recalibrate:

NoLoPhe, 2022-04-09:
## Basic Description

### Basic Description

#### Key features

Key features | Description
--- | ---
Outputs fused sensor data | Quaternion, Euler angles, Rotation vector, Linear acceleration, Gravity, Heading
3 sensors in one device | an advanced triaxial 16bit gyroscope, a versatile, leading edge triaxial 14bit accelerometer and a full performance geomagnetic sensor
Small package | LGA package 28 pins. Footprint 3.8 x 5.2 mm², height 1.13 mm²
Power Management | Intelligent Power Management: normal, low power and suspend mode available
Common voltage supplies | VDD voltage range: 2.4V to 3.6V
Digital interface | HID-I2C (Windows 8 compatible), I²C, UART VDDIO voltage range: 1.7V to 3.6V
Consumer electronics suite | MSL1, RoHS compliant, halogen-free. Operating temperature: -40°C - 85°C

NoLoPhe, 2022-04-10:
模块内部自带主动地磁校准系统，该系统不需要用户发送任何指令，该系统在后台自动采集一段时间内地磁场数据，并做分析比较，剔除异常数据，一旦数据足够，就会尝试地磁校准。

The module has its own active geomagnetic calibration system. The system does not require users to send any instructions. The system automatically collects magnetic field data for a period of time in the background, analyzes and compares it, and eliminates abnormal data. Once the data is sufficient, it will try geomagnetic calibration.

NoLoPhe, 2022-04-10:
所以，当使用9轴模式时，不需要用户任何干预即可完成地磁校准。

Therefore, when using the 9-axis mode, the geomagnetic calibration can be completed without any user intervention.

NoLoPhe, 2022-04-10:
但是模块仍然提供接口来让用户检查当前校准状态。自动校准的前提是需要有足够程度姿态机动(模块姿态的变化)并且维持一定时间，内部校准系统才能搜集不同姿态下的地磁场信息，从而完成校准，静止状态下是无法进行地磁校准的。

However, the module still provides an interface for the user to check the current calibration status. The premise of automatic calibration is to have sufficient degree of attitude maneuver (change of module attitude) and maintain a certain period of time, so that the internal calibration system can collect the magnetic field information under different attitudes to complete the calibration. Geomagnetic calibration cannot be performed in a static state.

NoLoPhe, 2022-04-10:
模块内部自带主动地磁校准系统，该系统不需要用户发送任何指令，该系统在后台自动采集一段时间内地磁场数据，并做分析比较，剔除异常数据，一旦数据足够，就会尝试地磁校准。所以，当使用9轴模式时，不需要用户任何干预即可完成地磁校准。但是模块仍然提供接口来让用户检查当前校准状态。自动校准的前提是需要有足够程度姿态机动(模块姿态的变化)并且维持一定时间，内部校准系统才能搜集不同姿态下的地磁场信息，从而完成校准，静止状态下是无法进行地磁校准的。

The module has its own active geomagnetic calibration system. The system does not require users to send any instructions. The system automatically collects magnetic field data for a period of time in the background, analyzes and compares it, and eliminates abnormal data. Once the data is sufficient, it will try geomagnetic calibration. Therefore, when using the 9-axis mode, the geomagnetic calibration can be completed without any user intervention. But the module still provides an interface for the user to check the current calibration status. The premise of automatic calibration is to have sufficient degree of attitude maneuver (change of module attitude) and maintain it for a certain period of time, so that the internal calibration system can collect the geomagnetic field information under different attitudes to complete the calibration. Geomagnetic calibration cannot be performed in a static state.

NoLoPhe, 2022-04-10:
首当首次使用模块并且需要使用9轴模式时，应进行如下校准操作：

When the module is used for the first time with the 9-axis mode, the following calibration operations should be performed:

NoLoPhe, 2022-04-10:
1. 检查周围是否磁干净：室内，实验室桌子旁，大型铁/刚框架结构附近。都属于常见的干扰区域。建议将模块拿到室外空旷处，即使没有条件拿到室外，尽量将模块远离(>0.5m)实验室桌子/电脑等容易产生干扰的物体。

1. Check whether the surroundings are clean: indoors, next to the laboratory table, near the large iron/steel frame structure. All belong to common interference areas. It is recommended to take the module to an outdoor open place. Even if there is no condition to take it outdoors, try to keep the module away from (>0.5m) laboratory desks/computers and other objects that are prone to interference.

1. Check if the surroundings are magnetically clean: indoors, next to a laboratory table, near large iron/rigid frame structures. All belong to the common interference area. It is recommended to take the module to an open space outdoors. Even if there is no condition to take it outdoors, try to keep the module away from (>0.5m) objects that are prone to interference such as laboratory desks/computers.

NoLoPhe, 2022-04-10:
在尽量小范围内(位置不动，只是旋转)，缓慢的让模块旋转，让模块经历尽量多的姿态位置(每个轴至少都旋转360°，持续约1分钟)。一般情况下即可完成校准。如果始终没能成功校准模块，说明周围地磁场干扰比较大。

In the smallest possible range (the position does not move, just rotates), slowly rotate the module, and let the module experience as many attitude positions as possible (each axis rotates at least 360° for about 1 minute). Under normal circumstances, the calibration can be completed. If the module is never successfully calibrated, it means that the surrounding geomagnetic field interference is relatively large.

In the smallest possible range (the position does not move, just rotate), slowly rotate the module, and let the module go through as many pose positions as possible (rotate each axis at least 360° for about 1 minute). In general, the calibration can be completed. If the module has not been successfully calibrated, it means that the surrounding geomagnetic field interference is relatively large.

NoLoPhe, 2022-04-10:
3. 校准的成功与否可用AT指令来查看：发送AT+INFO=HSI 指令，模块会打印当前地磁校准系统状态:

3. The success of the calibration can be checked by using the AT command: send the AT+INFO=HSI command, and the module will print the current status of the magnetometer calibration system:

NoLoPhe, 2022-04-10:
只需关心fiterr一项：0.03以下说明校准结果已经足够好。如果fiterr始终>0.1，说明地磁干扰很大，需要再次校准以期得到更好的校准结果。拟合残差会随着时间缓慢增长。

Just care about **fiterr**: 0.03 or less indicates that the calibration result is good enough. If **fiterr** is always > 0.1, it means that the geomagnetic interference is very large, and it needs to be calibrated again to get better calibration results. ~~拟合残差会随着时间缓慢增长。~~

NoLoPhe, 2022-04-10:
虽然地磁参数估计可以在线自动采集数据，自动的动态拟合地磁校准参数。但是如果周围地磁环境改变(比如需要到另外房间或者室内室外切换，或者是模块被安装/焊接到了新的环境中)，还需重复执行1-3.

Although geomagnetic parameter estimation can automatically collect data online, geomagnetic calibration parameters can be dynamically fitted automatically. However, if the surrounding geomagnetic environment changes (for example, you need to switch to another room or indoor and outdoor, or the module is installed/welded to a new environment), you need to repeat 1-3.

Although the geomagnetic parameter estimation can automatically collect data online, the geomagnetic calibration parameters are automatically and dynamically fitted. However, if the surrounding geomagnetic environment changes (for example, it needs to switch to another room or indoor and outdoor, or the module is installed/welded to a new environment), it is necessary to repeat 1-3.

NoLoPhe, 2022-04-10:
关于地磁干扰的更多知识

More about magnetic interference

NoLoPhe, 2022-04-10:
种类

定义 

典型干扰源 影响 

措施

type

Definition

Typical Interference Source Effects

measure

Types

definition

Influence of typical interference sources

measures

NoLoPhe, 2022-04-10:
空间磁场干扰(Distortions that do not move with sensor)

Space magnetic field interference (Distortions that do not move with sensor)

NoLoPhe, 2022-04-10:
## Hi229 Orientation

<p align="left">[image]</p>

The vector system uses the Front-Left-Up (FLU) right-handed coordinate system. The geographic coordinate system uses the North-West-Ultra (NWU) coordinate system. The Euler angle rotation order is: Z-Y-X (turn the Z axis first, then the Y axis, and finally the X axis) rotation sequence. The specific definitions are as follows:

- Rotation around the Z-axis: also called Yaw or (pronounced: Psi). Range: -180°- 180°
- Rotation around the Y-axis: also called Pitch or (pronounced: Theta). Range: -90°-90°
- Rotation around the X-axis: also called Roll or (pronounced: Phi). Range: -180°-180°

If you think of the module as an aircraft. The X-axis should be considered the direction of the machine head. When the sensor frame coincides with the inertial frame, the ideal output of Euler angles is: Pitch = 0°, Roll = 0°, Yaw = 0°

NoLoPhe, 2022-04-10:
## Packaging Information

### Package Outline

<p align="left">[image]</p>

Symbol | Minimum value | Typical value | Maximum value | Unit
--- | --- | --- | --- | ---
A1 | - | 11 | - | mm
B | - | 11 | - | mm
D | - | 12 | - | mm
E | - | 12 | - | mm
H | 2.5 | 2.6 | 2.7 | mm
a | - | 1.5 | - | mm
b | - | 0.9 | - | mm
c | - | 1 | - | mm
e | - | 1.27 | - | mm
f | - | 1 | - | mm

NoLoPhe, 2022-04-10:
安装及焊接
### ~~Mounting and Soldering~~ Soldering Guidelines

1. 安装位置请远离PCB容易形变点，尽量远离PCB边缘(>30mm)，远离PCB定位螺丝孔(>10mm)等。
1. The installation position should be away from the easy deformation point of the PCB, try to stay away from the edge of the PCB (>30mm), and away from the PCB positioning screw hole (>10mm).

2. 安装位置请远离强磁设备，如电机，喇叭等强磁器件。
2. The installation position should be away from strong magnetic devices, such as motors, speakers and other strong magnetic devices.

3. 组装好的PCB不得使用超声波清洁仪进行清洁。
3. The assembled PCB must not be cleaned with an ultrasonic cleaner.

4. 本产品不可使用塑封或喷涂三防漆，喷漆或塑封会造成传感器应力改变进而影响性能。
4. This product cannot be plastic-sealed or sprayed with conformal paint. Spraying or plastic-sealing will cause the sensor stress to change and affect its performance.

5. 推荐回流焊的炉温曲线图如下：
5. The recommended oven temperature graph for reflow soldering is as follows:

注意
Notice

回流焊最后阶段需要自然冷却，不能开启炉子强制风冷却，否则严重影响产品性能。
The final stage of reflow soldering requires natural cooling, and the furnace cannot be turned on for forced air cooling, otherwise the product performance will be seriously affected.

NoLoPhe, 2022-04-10:
## Getting Started with Hi229

模組與PC機連接
Connect the module to the PC

建議使用評估板與PC機進行連接，評估板板載USB供電及USB轉序列埠功能，可以方便的配合PC機上的評估軟體進行性能測試。具體請參見附錄中的評估板一節。
It is recommended to use the evaluation board to connect to the PC. The evaluation board has on-board USB power supply and USB to serial port functions, which can be conveniently used with the evaluation software on the PC for performance testing. See the Evaluation Boards section in the appendix for details.

模組與MCU進行連接
Connect the module to the MCU

模組與MCU通過TTL電平的序列埠進行連接，建議模組的RST引腳建議接到MCU的GPIO上。方便MCU強制復位模組。
The module and the MCU are connected through the serial port of TTL level. It is recommended that the RST pin of the module be connected to the GPIO of the MCU. It is convenient for MCU to force reset the module.

NoLoPhe, 2022-04-10:
## Getting Started with Hi229

模組與PC機連接
Connect the module to the PC

建議使用評估板與PC機進行連接，評估板板載USB供電及USB轉序列埠功能，可以方便的配合PC機上的評估軟體進行性能測試。具體請參見附錄中的評估板一節。
It is recommended to use the evaluation board to connect to the PC. The evaluation board has on-board USB power supply and USB to serial port functions, which can be conveniently used with the evaluation software on the PC for performance testing. See the Evaluation Boards section in the appendix for details.

模組與MCU進行連接
Connect the module to the MCU

模組與MCU通過TTL電平的序列埠進行連接，建議模組的RST引腳建議接到MCU的GPIO上。方便MCU強制復位模組。
The module and the MCU are connected through the serial port of TTL level. It is recommended that the RST pin of the module be connected to the GPIO of the MCU. It is convenient for MCU to force reset the module.

<p align="left">[image]</p>

注意
Notice

1. 如不使用同步輸入(SYNC_IN) 和同步輸出功能(SYNC_OUT) 可不接SYNC_IN 和 SYNC_OUT。
1. If the synchronization input (SYNC_IN) and synchronization output functions (SYNC_OUT) are not used, SYNC_IN and SYNC_OUT can be disconnected.

2. 120歐電阻的作用是為了方便調試，並防止MCU和模組電平不匹配，可以去掉，建議保留。
2. The function of the 120 ohm resistor is to facilitate debugging and prevent the level mismatch between the MCU and the module.  It can be removed. It is recommended to keep it.

3. ~~VCC~~ VDD的電壓範圍具體參加手冊說明
3. Please refer to the manual for the voltage range of ~~VCC~~ VDD

4. 模組內置上電復位電路，RST可以不接，但是建議接到主機一個GPIO上來實現軟體復位。
4. The module has built-in power-on reset circuit, RST can be disconnected, but it is recommended to connect RST to a GPIO of the host to realize software reset.

NoLoPhe, 2022-04-10:
### Reading/Writing the Hi229

序列埠通訊協議
Serial port communication protocol
 
模組上電後，默認按出廠幀率(通常為100)輸出幀數據，幀格式如下：
After the module is powered on, the frame data is output at the factory frame rate (usually 100) by default. The frame format is as follows:

``` 
Serial data frame structure
<frame header (0x5A)><frame type (0xA5)><length><CRC check><data field>
 ```
域名稱 Domain name
值 Value
長度(字節) Length (bytes)
說明 Description

Domain name | Value | Length (bytes) | Description
--- | --- | --- | ---
PRE | 0x5A | 1 | 固定為0x5A Fixed to 0x5A
TYPE | 0xA5 | 1 | 固定為0xA5 Fixed to 0xA5
LEN | 1-512 | 2 | 幀中數據域的長度，低字節在前。長度表示數據域(PAYLOAD)的長度，不包含 PRE , TYPE , LEN , CRC 字段。
| | | | The length of the data field in the frame, the low byte comes first. The length represents the length of the data field (PAYLOAD) and does not contain the PRE, TYPE, LEN, CRC fields.
CRC | - | 2 | 除CRC 本身外其餘所有字段( PRE,TYPE,LEN, PAYLOAD )幀數據的16 位CRC 校驗和。LSB(低字節在前)
| | | | Except for the CRC itself, the 16-bit CRC checksum of the frame data for all other fields (PRE, TYPE, LEN, PAYLOAD ). LSB (low byte first)
PAYLOAD | - | 1-512 | 一幀攜帶的數據。PAYLOAD域 由若干個子數據包組成。每個數據包包含數據包標籤和數據兩部分。標籤決定了數據的類型及長度。
| | | | The data carried in one frame. The PAYLOAD field consists of several sub-packets.  Each data packet contains two parts: the data packet label and the data. The label determines the type and length of the data.

NoLoPhe, 2022-04-10:
### Hi229 Configuration

數據包
Data package

數據包總覽
Data package overview

數據包標籤(DATA_ID) Data packet label (DATA_ID)
數據包長度(包含標籤1字節) Packet length (including label 1 byte)
名稱 Name
備註 Remarks

Record ID | Length (bytes) | Name | Condition
--- | --- | --- | ---
0x90 | 2 | 用戶ID User ID
0xA0 | 7 | 加速度 Acceleration
0xB0 | 7 | 角速度 Angular velocity
0xC0 | 7 | 磁場強度 Magnetic Field Strength
0xD0 | 7 | 歐拉角 Euler Angles
0xD1 | 17 | 四元數 Quaternion
0xF0 | 5 | 氣壓 Air Pressure | 輸出0 Output 0
0x91 | 76 | IMUSOL(IMU數據集合) IMUSOL (IMU data set) | 推薦使用 Recommended

NoLoPhe, 2022-04-11:
### Hi229 Configuration

Record ID | Length (bytes) | Name | Remark
--- | --- | --- | ---
0x90 | 2 | 用戶ID User ID
0xA0 | 7 | 加速度 Acceleration
0xB0 | 7 | 角速度 Angular velocity
0xC0 | 7 | 磁場強度 Magnetic Field Strength
0xD0 | 7 | 歐拉角 Euler Angles
0xD1 | 17 | 四元數 Quaternion
0xF0 | 5 | 氣壓 Air Pressure | 輸出0 Output 0
0x91 | 76 | IMUSOL(IMU數據集合) IMUSOL (IMU data set) | 推薦使用 Recommended

### Hi229 Reports

Product support data package list
A total of 2 bytes, the ID set by the user.

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0x90
1 | uint8_t | 1 | - | User ID

#### 0xA0 (Acceleration)

A total of 7 bytes, LSB. Output the raw acceleration of the sensor

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0xA0
1 | int16_t | 2 | 0.001G (1G = 1 gravitational acceleration) | X-axis acceleration
3 | int16_t | 2 | 0.001G | Y-axis acceleration
5 | int16_t | 2 | 0.001G | Z-axis acceleration

#### 0xB0 (Angular velocity)

A total of 7 bytes, LSB. Output the raw angular velocity of the sensor

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0xB0
1 | int16_t | 2 | 0.1°/s | X-axis angular velocity
3 | int16_t | 2 | 0.1°/s | Y-axis angular velocity
5 | int16_t | 2 | 0.1°/s | Z-axis angular velocity

#### 0xC0 (Magnetic Field Strength)

A total of 7 bytes, LSB. Output the original magnetic field strength of the sensor

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0xC0
1 | int16_t | 2 | 0.001Gauss | X-axis magnetic field strength
3 | int16_t | 2 | 0.001Gauss | Y-axis magnetic field strength
5 | int16_t | 2 | 0.001Gauss | Z-axis magnetic field strength

#### 0xD0 (Euler Angles)

A total of 7 bytes, LSB. The format is int16, a total of 3 values, each axis occupies 2 bytes, and the order is Pitch/Roll/Yaw. Roll, Pitch are the value obtained by multiplying the physical value by 100, and Yaw is the value obtained by multiplying it by 10.

Example: When the received Yaw = 100, it means that the heading angle is 10°

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0xD0
1 | int16_t | 2 | 0.01° | Pitch (pitch angle)
3 | int16_t | 2 | 0.01° | Roll (roll angle)
5 | int16_t | 2 | 0.1° | Yaw (heading angle)

#### 0xD1 (Quaternion)

A total of 17 bytes, LSB. The format is float, a total of 4 values, the order is: W X Y Z. Each value occupies 4 bytes (float), and the entire quaternion is 4 floats, LSB.

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0xD1
1 | float | 4 | - | W
5 | float | 4 | - | X
9 | float | 4 | - | Y
13 | float | 4 | - | Z

#### 0xF0 (Air Pressure)

A total of 5 bytes, the format is float. (Only for products with air pressure sensor)

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0xF0
1 | float | 4 | Pa | Atmospheric pressure

#### 0x91 IMUSOL (IMU data set)

A total of 76 bytes, the newly added data packet is used to replace A0, B0, C0, D0, D1 and other data packets. Integrate the sensor raw output of the IMU and the attitude solution data.

Byte | Type | Size | Unit | Description
--- | --- | --- | --- | ---
0 | uint8_t | 1 | - | Data packet label: 0x91
1 | uint8_t | 1 | - | ID
2 | - | 6 | - | Reserved
8 | uint32_t | 4 | ms | Timestamp information, accumulated from the system boot, increases by 1 per millisecond
12 | float | 12 | (1G = 1 acceleration of gravity) | Acceleration on the X, Y, and Z axis, note that the unit is different from 0xA0
24 | float | 12 | deg/s | the angular velocity of the X, Y, and Z axis, note that the unit is different from 0xB0
36 | float | 12 | uT | the magnetic field strength of X, Y, and Z axis (Hi229 support, note that the unit is different from 0xC0)
48 | float | 12 | deg | A set of Euler angles of nodes, in the order: Roll angle (Roll), Pitch angle (Pitch), heading angle (Yaw) (note that the order and unit are different from the 0xD0 data packet)
60 | float | 16 | - | WXYZ Set of node quaternions, the order is WXYZ

NoLoPhe, 2022-04-11:
出廠默認數據包
Factory default data package

出廠默認一幀中攜帶數據包數據定義如下：
The factory default definition of packet data carried in a frame is as follows:

產品 
Product 
 
默認輸出數據包
Default output data packet

Product | Default output data packet
--- | ---
Hi229 | 0x91

NoLoPhe, 2022-04-11:
數據幀配置為 0x91 數據包
The data frame is configured as a 0x91 data packet

使用序列埠助手採樣一幀數據,共82字節, 前6字節為幀頭, 長度和CRC校驗值。剩餘76字節為數據域。假設數據接收到C語言陣列 buf 中。如下所示:
Use the serial port assistant to sample a frame of data, a total of 82 bytes, the first 6 bytes are the frame header, length and CRC check value. The remaining 76 bytes are data fields. Suppose the data is received into the C language array buf. As follows:

- 第一步：判斷幀頭，得到數據域長度和幀CRC：
- Step 1: Determine the frame header, get the data field length and frame CRC:

幀頭: 5A A5
Frame header: 5A A5

幀數據域長度: 4C 00 : (0x00<<8) + 0x4C = 76
Frame data field length: 4C 00 : (0x00<<8) + 0x4C = 76

幀CRC校驗值: 6C 51 :(0x51<<8) + 0x6C = 0x516C
Frame CRC check value: 6C 51 :(0x51<<8) + 0x6C = 0x516C

- 第二步： 校驗CRC
- Step 2: Check CRC

得到CRC值為0x516C. 幀CRC校驗通過。
The CRC value obtained is 0x516C. The frame CRC check is passed.

- 第三步：接收數據
- Step 3: Receive data

從 0x91 開始為數據包的數據域。在C語言中可以定義結構體來方便的讀取數據：
The data field of the data packet starts from 0x91. In C language, you can define a structure to easily read data:

定義0x91數據包結構體如下：
Define the 0x91 packet structure as follows:

__packed 為編譯器關鍵字(Keil下)，表示結構體按字節緊對齊，結構體每一個元素一一對應0x91數據包的結構定義。接收數據時將接收到的陣列直接memcpy到結構體即可：(注意定義結構體時必須4字節對齊), 其中 buf 指向幀頭, buf[6] 指向幀中數據域 。
 __packed is the compiler keyword (Keil), which means that the structure is tightly aligned by bytes, and each element of the structure corresponds to the structure definition of the 0x91 data packet. When receiving data, `memcpy` the received array directly to the structure: (notice that 4 bytes must be aligned when defining the structure), where `buf` points to the frame header and `buf [6]` points to the data field in the frame. 

最後得到dat數據結果：
Finally get the dat data result：

NoLoPhe, 2022-04-11:
#### Factory default data package

The factory default definition of packet data carried in a frame is as follows:

Product | Default output data packet
--- | ---
Hi229 | 0x91

#### Example of data frame structure
 
##### The data frame is configured as 0x90, 0xA0,0xB0,0xC0,0xD0, 0xF0 data packets

Use the serial port assistant to sample a frame of data, a total of 41 bytes, the first 6 bytes are the frame header, length and CRC check value. The remaining 35 bytes are data fields. Suppose the data is received into the C language array `buf` . As follows:

5A A5 23 00 FD 61 **90** 00 **A0** 55 02 3D 01 E2 02 **B0** FE FF 17 00 44 00 **C0** 80 FF 60 FF 32 FF **D0** 64 F2 6C 0E BB 01 **F0** 00 00 00 00

- The first step: determine the frame header, get the data field length and frame CRC:

Frame header: 5A A5

Frame data field length: 23 00 : (0x00<<8) + 0x23 = 35

Frame CRC check value: FD 61 :(0x61<<8) + 0xFD = 0x61FD

- Step 2: Check CRC

```
    uint16_t payload_len; 
    uint16_t crc;

    crc = 0;
    payload_len = buf[2] + (buf[3] << 8);

    /* calulate 5A A5 and LEN filed crc */
    crc16_update(&crc, buf, 4);

    /* calulate payload crc */
    crc16_update(&crc, buf + 6, payload_len);
```

The obtained CRC value is 0x61FD, which is the same as the CRC value carried by the frame, and the frame CRC check is passed. 

- Step 3: Receive data

90 00: ID data packet, 0x90 is the data packet label, ID = 0x00. 

A0 55 02 3D 01 E2 02: Acceleration data packet, 0xA0 is the data packet label, and the three-axis acceleration is：

X-axis acceleration = (int16_t) ((0x02<<8) +0x55) =597 (unit is mG)
Y-axis acceleration = (int16_t) ((0x01<<8)+0x3D) =317
Z-axis acceleration = (int16_t) ((0x02<<8) +0xE2) =738

B0 FE FF 17 00 44 00: Angular velocity data packet, 0xB0 is the data packet label, and the three-axis angular velocity is：

X-axis angular velocity = (int16_t)((0xFF<<8)+ 0xFE) = -2 (unit is 0.1°/s)
Y-axis angular velocity = (int16_t)((0x00<<8)+ 0x17) = 23
Z-axis angular velocity = (int16_t)((0x00<<8)+ 0x44) = 68

C0 80 FF 60 FF 32 FF: Magnetic field data packet, 0xC0 is the data packet label, and the three-axis magnetic field is：

X-axis angular velocity = (int16_t)((0xFF<<8)+ 0x80) = -128 (unit is 0.001GAUSS)
Y-axis angular velocity = (int16_t)((0xFF<<8)+ 0x60) = -160
Z-axis angular velocity = (int16_t)((0xFF<<8)+ 0x32) = -206

D0 64 F2 6C 0E BB 01 Euler angle data packet, 0xD0 is the data packet label

Pitch= (int16_t)((0xF2<<8)+ 0x64) / 100 = -3484 / 100 = -34.84 °
Roll= (int16_t)((0x0E<<8)+ 0x6C) / 100 = 3692 / 100 = 36.92°
Yaw = (int16_t)((0x01<<8)+ 0xBB) / 10 = 443 /10 = 44.3°

F0 00 00 00 00 air pressure data packet, 0xF0 is the data packet label

```
    float prs;
    prs = memcpy(&prs, &buf[37], 4);
```

Finally get the result：

```
id : 0
acc(G) : 0.597 0.317 0.738
gyr(deg/s) : -0.200 2.300 6.800
mag(uT) : -12.800 -16.000 -20.600
eul(R/P/Y) : 36.920 -34.840 44.300
```

NoLoPhe, 2022-04-13:
陀螺仪
Gyroscope

参数 Parameter | 值 Value | 备注 ~~Remarks~~ Condition
--- | --- | ---
测量范围 Measuring range Phạm vi đo | ±2000°/s
分辨率 Resolution Độ phân giải | 0.01°/s
内部采样频率 Internal sampling frequency Tần suất lấy mẫu nội bộ | 1KHz
零偏稳定性 Zero bias stability Độ ổn định không thiên vị | 8°/h | @25°C,1σ
零偏重复性 Zero-bias repeatability Độ lặp lại không thiên vị | 0.12°/s | @25°C,1σ
非正交误差 Non-orthogonal errors (including position/displacement errors, pointing/azimuth errors, installation errors, etc.) Lỗi không trực giao (bao gồm lỗi vị trí / dịch chuyển, lỗi trỏ / phương vị, lỗi cài đặt, v.v.) | ±0.1% | @25°C,1σ
随机游走 Random walk | 0.6 | @25°C,1σ
刻度非线性度 Scale nonlinearity Quy mô phi tuyến tính | ±0.1% | 满量程时(最大) At full scale (maximum) 
刻度系数误差 Scale factor error Lỗi hệ số thang đo (Lỗi hệ số tỉ lệ) [Link](https://zhuanlan.zhihu.com/p/352539057) | ±0.4% | 出厂前校准后 After factory calibration
加速度敏感性 Acceleration sensitivity | 0.1°/s/g

NoLoPhe, 2022-04-13:
加速度计
Accelerometer

参数 Parameter | 值 Value | 备注 ~~Remarks~~ Condition
--- | --- | ---
测量范围 Measurement range | ±8G (1G = 1x 重力加速度 ) ±8G (1G = 1x acceleration of gravity) 
分辨率 Resolution | 1uG
内部采样频率 Internal sampling frequency | 1KHz
零偏稳定性 Zero-bias stability | 60uG @25°C,1
零偏重复性 Zero-bias repeatability | 4.8mG @25°C,1
非正交误差 Non-orthogonal | ±0.1%
随机游走 Random walk | 0.08 | @25°C,1
刻度系数误差 Scale factor error | ±0.3% | (满量程时) 出厂前校准后 (at full scale) after factory calibration
全温范围温度变化 Operating temperature | 2mg | -20 - 85°

[11 more replies]

closed 2022-04-14 by NoLoPhe