SlimeVRX/SlimeVRX #28: [paper] PIP

SlimeVRX, 2022-07-09:
Abstract

Motion capture from sparse inertial sensors has shown great potential compared to image-based approaches since occlusions do not lead to a reduced tracking quality and the recording space is not restricted to be within the viewing frustum of the camera. However, capturing the motion and global position only from a sparse set of inertial sensors is inherently ambiguous and challenging. In consequence, recent state-of-the-art methods can barely handle very long period motions, and unrealistic artifacts are common due to the unawareness of physical constraints. To this end, we present the first method which combines a neural kinematics estimator and a physics-aware motion optimizer to track body motions with only 6 inertial sensors. The kinematics module first regresses the motion status as a reference, and then the physics module refines the motion to satisfy the physical constraints. Experiments demonstrate a clear improvement over the state of the art in terms of capture accuracy, temporal stability, and physical correctness.

SlimeVRX, 2022-07-09:
Khả năng chụp chuyển động từ các cảm biến quán tính thưa thớt đã cho thấy tiềm năng lớn so với các phương pháp tiếp cận dựa trên hình ảnh vì hiện tượng khớp cắn không dẫn đến giảm chất lượng theo dõi và không gian ghi không bị giới hạn trong tầm nhìn của máy ảnh. Tuy nhiên, việc ghi lại chuyển động và vị trí toàn cầu chỉ từ một bộ cảm biến quán tính thưa thớt vốn dĩ rất mơ hồ và đầy thách thức. Do đó, các phương pháp hiện đại gần đây hầu như không thể xử lý các chuyển động trong thời gian rất dài, và các hiện vật phi thực tế thường gặp do không nhận thức được các ràng buộc vật lý. Để đạt được mục đích này, chúng tôi trình bày phương pháp đầu tiên kết hợp công cụ ước tính chuyển động thần kinh và công cụ tối ưu hóa chuyển động nhận biết vật lý để theo dõi chuyển động của cơ thể chỉ với 6 cảm biến quán tính. Đầu tiên mô-đun động học hồi quy trạng thái chuyển động như một tham chiếu, sau đó mô-đun vật lý tinh chỉnh chuyển động để thỏa mãn các ràng buộc vật lý. Các thử nghiệm cho thấy sự cải thiện rõ ràng so với hiện đại về độ chính xác khi chụp, độ ổn định theo thời gian và độ chính xác vật lý.

NoLoPhe, 2022-07-10:
2. Related Work:

Human motion capture (mocap) has a long research history. Many works have been devoted to this topic, which can be mainly categorized into optical, inertial, and hybrid approaches. Since our method only requires IMU measurements as input, we do not discuss purely image-based approaches [[](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark61)15[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark62)16[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark70)24[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark73)27[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark85)40[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark93)48[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark103)58]. Here, we focus on hybrid and inertial mocap solutions, and the previous efforts on the physical plausibility of human motion.

Optical-inertial Hybrid Motion Capture. [As image-based mocap solutions suffer from occlusions, fusing images with IMUs, which aims at achieving more robust motion tracking, has recently attracted much attention. This can be achieved by either energy-based optimization [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark68)22[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark82)36[–](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark83)38[,](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark88)43[,](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark110)65[] which optimizes human pose to fit both image features and inertia measurements, or feature-based estimation [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark59)13[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark107)62[] which regresses human pose from the combined features derived from images and IMUs. Zhang et al. [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark131)87[] propose to exploit IMUs in the 2D pose estimation by fusing the image features of each pair of joints linked by the IMUs. Some works fuse IMUs with depth images [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark64)18[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark69)23[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark135)90[] or optical markers [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark47)1] to perform human motion/performance capture. Nevertheless, these methods are still substantially limited under low light conditions and heavy occlusions, and require the actor to move within the viewing frustum of the camera. Our method requires no visual input, and thus is free from these limitations.

Motion Capture from Inertial Sensors. [Inertial mocap approaches do not suffer from occlusions or restricted moving space. Commercial solutions [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark80)34[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark116)71[] and the extended work [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark60)14[] rely on 17 IMUs to perform motion capture. They usually require the actor to wear a tight suit with densely bounded IMUs, which is inconvenient, intrusive, and obstructive. It is clear that having a reduced set of IMUs on the body is preferable. However, motion capture from sparse inertial sensors is very ambiguous and challenging. Some works [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark77)31[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark104)59[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark109)64[] leverage ultrasonic sensors for additional position information to resolve some ambiguities, but the use of distance sensors limits the recording range. Early purely-inertial works [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark92)47[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark102)57[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark105)60[] use sparse accelerometers to reconstruct human pose by database search. Schwarz et al. [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark95)50[] use sparse orientation measurements to perform person-specific pose estimation. To improve the accuracy, recent works [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark58)12[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark66)20[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark89)44[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark111)66[, ](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark118)73[] leverage both acceleration and orientation measurements. Marcard et al. [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark111)66[] present an offline method for human motion capture from only 6 IMUs, which achieves promising accuracy. Huang et al. [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark66)20] propose the first deep learning method, which uses a bidirectional recurrent neural network (biRNN) to estimate the human pose from 6 IMUs in real-time. However, their method does not allow to locate the person in the 3D space, i.e., [the root translation is not estimated. The current state-of-the-art method, TransPose [](https://github.com/SlimeVRX/SlimeVRX/issues/27#bookmark118)73], introduces the first real-time pose and translation estimation framework, which achieves an accurate capture quality while also only using 6 IMUs. However, all of these works have a non-negligible delay due to the inherent need of future information, cannot stably capture ambiguous poses, and has many non-physical artifacts such as jitter and foot-sliding. In contrast, our method does not rely on any future information and is free from the delay while even achieving higher accuracy. In addition, we are the first to combine physics-based motion optimization with sparse inertial motion capture, and we show that such a carefully orchestrated design significantly improves the physical correctness of the motion.

NoLoPhe, 2022-07-10:
Công việc liên quan:

Chụp chuyển động của con người (mocap) đã có lịch sử nghiên cứu lâu đời. Nhiều công trình đã được dành cho chủ đề này, chủ yếu có thể được phân loại thành các phương pháp tiếp cận quang học, quán tính và lai. Vì phương pháp của chúng tôi chỉ yêu cầu các phép đo IMU làm đầu vào, nên chúng tôi không thảo luận về các phương pháp tiếp cận dựa trên hình ảnh thuần túy [15, 16, 24, 27, 40, 48, 58]. Ở đây, chúng tôi tập trung vào các giải pháp mocap lai và quán tính, và những nỗ lực trước đây về tính hợp lý vật lý của chuyển động của con người.

Chụp chuyển động lai quán tính quang học. [Khi các giải pháp mocap dựa trên hình ảnh bị hiện tượng ăn khớp, việc kết hợp hình ảnh với IMU, nhằm mục đích đạt được khả năng theo dõi chuyển động mạnh mẽ hơn, gần đây đã thu hút nhiều sự chú ý. Điều này có thể đạt được bằng cách tối ưu hóa dựa trên năng lượng 22, 36–38,43,65 [], tối ưu hóa tư thế người để phù hợp với cả các đặc điểm hình ảnh và phép đo quán tính hoặc ước tính dựa trên tính năng 13, 62 [] trong đó hồi quy tư thế của con người từ các tính năng kết hợp có nguồn gốc từ hình ảnh và IMU. Zhang và cộng sự. 87 [] đề xuất khai thác IMU trong ước tính tư thế 2D bằng cách kết hợp các đặc điểm hình ảnh của từng cặp khớp được liên kết bởi IMU. Một số hoạt động kết hợp IMU với hình ảnh độ sâu 18, 23, 90 [] hoặc điểm đánh dấu quang học 1] để thực hiện chụp chuyển động / hiệu suất của con người. Tuy nhiên, các phương pháp này vẫn còn hạn chế đáng kể trong điều kiện ánh sáng yếu và khớp cắn nặng, đồng thời yêu cầu diễn viên phải di chuyển trong tầm nhìn của máy ảnh. Phương pháp của chúng tôi không yêu cầu đầu vào trực quan và do đó không có những hạn chế này.

Chụp chuyển động từ cảm biến quán tính. [Phương pháp tiếp cận mocap quán tính không bị sai khớp cắn hoặc hạn chế không gian di chuyển. Các giải pháp thương mại 34, 71 [] và công việc mở rộng 14 [] dựa vào 17 IMU để thực hiện chụp chuyển động. Họ thường yêu cầu diễn viên mặc một bộ đồ bó sát với các IMU dày đặc, điều này gây bất tiện, xâm nhập và cản trở. Rõ ràng là nên có một bộ IMU giảm bớt trên cơ thể. Tuy nhiên, việc chụp chuyển động từ các cảm biến quán tính thưa thớt là rất mơ hồ và khó khăn. Một số công trình 31, 59, 64 [] sử dụng cảm biến siêu âm để có thêm thông tin vị trí nhằm giải quyết một số điểm mơ hồ, nhưng việc sử dụng cảm biến khoảng cách sẽ hạn chế phạm vi ghi. Các tác phẩm đầu tiên thuần túy quán tính 47, 57, 60 [] sử dụng gia tốc kế thưa thớt để tái tạo lại tư thế của con người bằng cách tìm kiếm cơ sở dữ liệu. Schwarz và cộng sự. 50 [] sử dụng các phép đo định hướng thưa thớt để thực hiện ước tính tư thế dành riêng cho từng người. Để cải thiện độ chính xác, các công trình gần đây 12, 20, 44, 66, 73 [] sử dụng cả phép đo gia tốc và định hướng. Marcard và cộng sự. 66 [] trình bày một phương pháp ngoại tuyến để chụp chuyển động của con người chỉ từ 6 IMU, đạt được độ chính xác đầy hứa hẹn. Huang và cộng sự. 20] đề xuất phương pháp học sâu đầu tiên, sử dụng mạng nơ-ron lặp lại hai chiều (biRNN) để ước tính tư thế con người từ 6 IMU trong thời gian thực. Tuy nhiên, phương pháp của họ không cho phép xác định vị trí của người đó trong không gian 3D, tức là [bản dịch gốc không được ước tính. Phương pháp hiện đại nhất, TransPose 73], giới thiệu khung ước lượng dịch và tư thế thời gian thực đầu tiên, giúp đạt được chất lượng chụp chính xác trong khi cũng chỉ sử dụng 6 IMU. Tuy nhiên, tất cả các tác phẩm này đều có độ trễ không đáng kể do nhu cầu thông tin trong tương lai vốn có, không thể chụp ổn định các tư thế mơ hồ và có nhiều hiện vật phi vật lý như rung và trượt chân. Ngược lại, phương pháp của chúng tôi không dựa vào bất kỳ thông tin nào trong tương lai và không bị chậm trễ trong khi thậm chí đạt được độ chính xác cao hơn. Ngoài ra, chúng tôi là người đầu tiên kết hợp tối ưu hóa chuyển động dựa trên vật lý với chụp chuyển động quán tính thưa thớt và chúng tôi cho thấy rằng một thiết kế được tổ chức cẩn thận như vậy sẽ cải thiện đáng kể độ chính xác vật lý của chuyển động.

NoLoPhe, 2022-07-10:
Công việc liên quan:

Chụp chuyển động của con người (mocap) đã có lịch sử nghiên cứu lâu đời. Nhiều nghiên cứu đã được dành cho chủ đề này, chủ yếu có thể được phân loại thành các phương pháp tiếp cận quang học, quán tính và lai. Vì phương pháp của chúng tôi chỉ yêu cầu các phép đo IMU làm đầu vào, nên chúng tôi không thảo luận về các phương pháp tiếp cận dựa trên hình ảnh thuần túy [15, 16, 24, 27, 40, 48, 58]. Ở đây, chúng tôi tập trung vào các giải pháp mocap lai và quán tính, và những nỗ lực trước đây về tính hợp lý vật lý của chuyển động của con người.

Chụp chuyển động lai quán tính quang học. Khi các giải pháp mocap dựa trên hình ảnh bị hiện tượng ăn khớp, việc kết hợp hình ảnh với IMU, nhằm mục đích đạt được khả năng theo dõi chuyển động mạnh mẽ hơn, gần đây đã thu hút nhiều sự chú ý. Điều này có thể đạt được bằng cách tối ưu hóa dựa trên năng lượng 22, 36–38,43,65 [], tối ưu hóa tư thế người để phù hợp với cả các đặc điểm hình ảnh và phép đo quán tính hoặc ước tính dựa trên tính năng 13, 62 [] trong đó hồi quy tư thế của con người từ các tính năng kết hợp có nguồn gốc từ hình ảnh và IMU. Zhang và cộng sự. 87 [] đề xuất khai thác IMU trong ước tính tư thế 2D bằng cách kết hợp các đặc điểm hình ảnh của từng cặp khớp được liên kết bởi IMU. Một số hoạt động kết hợp IMU với hình ảnh độ sâu 18, 23, 90 [] hoặc điểm đánh dấu quang học 1] để thực hiện chụp chuyển động / hiệu suất của con người. Tuy nhiên, các phương pháp này vẫn còn hạn chế đáng kể trong điều kiện ánh sáng yếu và khớp cắn nặng, đồng thời yêu cầu diễn viên phải di chuyển trong tầm nhìn của máy ảnh. Phương pháp của chúng tôi không yêu cầu đầu vào trực quan và do đó không có những hạn chế này.

Chụp chuyển động từ cảm biến quán tính. [Phương pháp tiếp cận mocap quán tính không bị sai khớp cắn hoặc hạn chế không gian di chuyển. Các giải pháp thương mại 34, 71 [] và công việc mở rộng 14 [] dựa vào 17 IMU để thực hiện chụp chuyển động. Họ thường yêu cầu diễn viên mặc một bộ đồ bó sát với các IMU dày đặc, điều này gây bất tiện, xâm nhập và cản trở. Rõ ràng là nên có một bộ IMU giảm bớt trên cơ thể. Tuy nhiên, việc chụp chuyển động từ các cảm biến quán tính thưa thớt là rất mơ hồ và khó khăn. Một số công trình 31, 59, 64 [] sử dụng cảm biến siêu âm để có thêm thông tin vị trí nhằm giải quyết một số điểm mơ hồ, nhưng việc sử dụng cảm biến khoảng cách sẽ hạn chế phạm vi ghi. Các tác phẩm đầu tiên thuần túy quán tính 47, 57, 60 [] sử dụng gia tốc kế thưa thớt để tái tạo lại tư thế của con người bằng cách tìm kiếm cơ sở dữ liệu. Schwarz và cộng sự. 50 [] sử dụng các phép đo định hướng thưa thớt để thực hiện ước tính tư thế dành riêng cho từng người. Để cải thiện độ chính xác, các công trình gần đây 12, 20, 44, 66, 73 [] sử dụng cả phép đo gia tốc và định hướng. Marcard và cộng sự. 66 [] trình bày một phương pháp ngoại tuyến để chụp chuyển động của con người chỉ từ 6 IMU, đạt được độ chính xác đầy hứa hẹn. Huang và cộng sự. 20] đề xuất phương pháp học sâu đầu tiên, sử dụng mạng nơ-ron lặp lại hai chiều (biRNN) để ước tính tư thế con người từ 6 IMU trong thời gian thực. Tuy nhiên, phương pháp của họ không cho phép xác định vị trí của người đó trong không gian 3D, tức là [bản dịch gốc không được ước tính. Phương pháp hiện đại nhất, TransPose 73], giới thiệu khung ước lượng dịch và tư thế thời gian thực đầu tiên, giúp đạt được chất lượng chụp chính xác trong khi cũng chỉ sử dụng 6 IMU. Tuy nhiên, tất cả các tác phẩm này đều có độ trễ không đáng kể do nhu cầu thông tin trong tương lai vốn có, không thể chụp ổn định các tư thế mơ hồ và có nhiều hiện vật phi vật lý như rung và trượt chân. Ngược lại, phương pháp của chúng tôi không dựa vào bất kỳ thông tin nào trong tương lai và không bị chậm trễ trong khi thậm chí đạt được độ chính xác cao hơn. Ngoài ra, chúng tôi là người đầu tiên kết hợp tối ưu hóa chuyển động dựa trên vật lý với chụp chuyển động quán tính thưa thớt và chúng tôi cho thấy rằng một thiết kế được tổ chức cẩn thận như vậy sẽ cải thiện đáng kể độ chính xác vật lý của chuyển động.

closed 2022-07-10 by NoLoPhe