SlimeVRX/SlimeVRX #29: [thuyết minh]

SlimeVRX, 2022-07-09:
Tóm tắt Abstract

Khả năng chụp chuyển động từ các cảm biến quán tính đã cho thấy tiềm năng lớn so với các phương pháp tiếp cận dựa trên hình ảnh vì không gặp phải những khó khan như không gian chuyển động phạm vi rộng và tắc nghẽn do các bộ phận cơ thể người bị che khuất. Các phương pháp hiện đại gần đây hầu như sử dụng các cảm biến quán tính chuyên dụng từ Xsens và Noitom chi phí rất cao dẫn đến việc triển khai thức tế rất khó khăn và khó phổ biến. Việc ghi lại chuyển động và vị trí toàn cầu chỉ từ một bộ cảm biến quán tính giá rẻ và chạy real-time trên CPU đầy thách thức. Để đạt được mục đích này, chúng tôi trình bày tái tạo lại kết quả của bài báo Transpose sử dụng 6 cảm biến quán tính giá rẻ Hi229 và tối ưu model neural network cho CPU đạt 24 FPS real-time. Các thử nghiệm cho thấy chúng tôi hoàn toàn có thể sử dụng cảm biến giá rẻ cho việc chụp chuyển động với độ chính xác và độ ổn định theo thời gian.

SlimeVRX, 2022-07-09:
Contents
1 Introduction 1
1.1 Background . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . 1
1.2 Motivation . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . 3
1.3 Objectives and Contributions . . . . . . . . . . . . . . . . . . . . . . . 3
1.4 Organization . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . 3

NoLoPhe, 2022-07-09:
Khả năng chụp chuyển động từ các cảm biến quán tính đã cho thấy tiềm năng lớn so với các phương pháp tiếp cận dựa trên hình ảnh vì không gặp phải những khó khăn như không gian chuyển động phạm vi rộng và tắc nghẽn do các bộ phận cơ thể người bị che khuất. Các phương pháp hiện đại gần đây hầu như sử dụng các cảm biến quán tính chuyên dụng từ Xsens và Noitom chi phí rất cao dẫn đến việc triển khai thức tế rất khó khăn và khó phổ biến. Việc ghi lại chuyển động và vị trí toàn cầu chỉ từ một bộ cảm biến quán tính giá rẻ và chạy real-time trên CPU đầy thách thức. Để đạt được mục đích này, chúng tôi trình bày tái tạo lại kết quả của bài báo Transpose sử dụng 6 cảm biến quán tính giá rẻ Hi229 và tối ưu model neural network cho CPU đạt 24 FPS real-time. Các thử nghiệm cho thấy chúng tôi hoàn toàn có thể sử dụng cảm biến giá rẻ cho việc chụp chuyển động với độ chính xác và độ ổn định theo thời gian.

SlimeVRX, 2022-07-09:
1 Introduction

1.1 Background 

Chụp chuyển động của con người (mocap), nhằm mục đích tái tạo lại các chuyển động cơ thể con người 3D, đóng một vai trò quan trọng trong các ứng dụng khác nhau như chơi game, thể thao, y học, VR / AR và sản xuất phim. Cho đến nay, các giải pháp mocap dựa trên tầm nhìn chiếm phần lớn trong chủ đề này. Một danh mục yêu cầu gắn các điểm đánh dấu quang học trên cơ thể người và sử dụng nhiều máy ảnh để theo dõi các điểm đánh dấu để ghi lại chuyển động. Các hệ thống dựa trên điểm đánh dấu như Vicon1 [được áp dụng rộng rãi và được coi là đủ chính xác cho việc sử dụng công nghiệp. Tuy nhiên, những cách tiếp cận như vậy yêu cầu cơ sở hạ tầng đắt tiền và các thiết bị xâm nhập, điều này khiến chúng không được mong muốn đối với cách sử dụng ở cấp độ người tiêu dùng. Gần đây, một danh mục khác tập trung vào ước tính tư thế sử dụng một vài máy ảnh RGB hoặc RGB-D Chen et al. Năm 2020; Habibie và cộng sự. Năm 2019; Mehta và cộng sự. Năm 2020; Tome và cộng sự. Năm 2018; Trumble và cộng sự. 2016; Xiang và cộng sự. Năm 2019]. Mặc dù các phương pháp này nhẹ hơn nhiều, nhưng chúng rất nhạy cảm với sự xuất hiện của con người vì các đặc điểm phân biệt cần được trích xuất từ ​​hình ảnh. Do đó, các phương pháp này thường hoạt động kém hiệu quả đối với quần áo không có họa tiết hoặc ánh sáng môi trường khó khăn. Hơn nữa, tất cả các phương pháp tiếp cận dựa trên thị lực đều bị tắc. Một vấn đề như vậy đôi khi có thể được giải quyết bằng cách đặt các camera dày đặc (điều này càng làm cho hệ thống trở nên nặng nề và tốn kém), nhưng nó thường không thực tế trong một số ứng dụng. Ví dụ, rất khó bố trí máy ảnh trong một phòng sinh hoạt chung có đầy đủ đồ đạc hoặc vật dụng, điều này có thể làm chủ thể bị che khuất từ ​​bất kỳ hướng nào. Một [hạn chế khác của phương pháp tiếp cận dựa trên tầm nhìn là những người biểu diễn thường bị hạn chế trong một không gian cố định. Đối với các hoạt động hàng ngày như đi bộ và chạy trong phạm vi lớn, cần phải có camera chuyển động được kiểm soát cẩn thận để ghi lại đủ thông tin chuyển động, điều này rất khó đạt được Xu et al. 2016]. Những nhược điểm này là những sai sót nghiêm trọng đối với nhiều ứng dụng và do đó dẫn đến khả năng sử dụng hạn chế của các giải pháp dựa trên tầm nhìn.

NoLoPhe, 2022-07-09:
1 Introduction

1.1 Background

Chụp chuyển động của con người (mocap), nhằm mục đích tái tạo lại các chuyển động cơ thể con người 3D, đóng một vai trò quan trọng trong các ứng dụng khác nhau như chơi game, thể thao, y học, VR / AR và sản xuất phim. Cho đến nay, các giải pháp mocap dựa trên tầm nhìn chiếm phần lớn trong chủ đề này. Một danh mục yêu cầu gắn các điểm đánh dấu quang học trên cơ thể người và sử dụng nhiều máy ảnh để theo dõi các điểm đánh dấu để ghi lại chuyển động. Các hệ thống dựa trên điểm đánh dấu như Vicon1 [được áp dụng rộng rãi và được coi là đủ chính xác cho việc sử dụng công nghiệp. Tuy nhiên, những cách tiếp cận như vậy yêu cầu cơ sở hạ tầng đắt tiền và các thiết bị xâm nhập, điều này khiến chúng không được mong muốn đối với cách sử dụng ở cấp độ người tiêu dùng. Gần đây, một danh mục khác tập trung vào ước tính tư thế sử dụng một vài máy ảnh RGB hoặc RGB-D Chen et al. Năm 2020; Habibie và cộng sự. Năm 2019; Mehta và cộng sự. Năm 2020; Tome và cộng sự. Năm 2018; Trumble và cộng sự. 2016; Xiang và cộng sự. Năm 2019]. Mặc dù các phương pháp này nhẹ hơn nhiều, nhưng chúng rất nhạy cảm với sự xuất hiện của con người vì các đặc điểm phân biệt cần được trích xuất từ ​​hình ảnh. Do đó, các phương pháp này thường hoạt động kém hiệu quả đối với quần áo không có họa tiết hoặc ánh sáng môi trường khó khăn. Hơn nữa, tất cả các phương pháp tiếp cận dựa trên thị lực đều bị tắc. Một vấn đề như vậy đôi khi có thể được giải quyết bằng cách đặt các camera dày đặc (điều này càng làm cho hệ thống trở nên nặng nề và tốn kém), nhưng nó thường không thực tế trong một số ứng dụng. Ví dụ, rất khó bố trí máy ảnh trong một phòng sinh hoạt chung có đầy đủ đồ đạc hoặc vật dụng, điều này có thể làm chủ thể bị che khuất từ ​​bất kỳ hướng nào. Một [hạn chế khác của phương pháp tiếp cận dựa trên tầm nhìn là những người biểu diễn thường bị hạn chế trong một không gian cố định. Đối với các hoạt động hàng ngày như đi bộ và chạy trong phạm vi lớn, cần phải có camera chuyển động được kiểm soát cẩn thận để ghi lại đủ thông tin chuyển động, điều này rất khó đạt được Xu et al. 2016]. Những nhược điểm này là những sai sót nghiêm trọng đối với nhiều ứng dụng và do đó dẫn đến khả năng sử dụng hạn chế của các giải pháp dựa trên tầm nhìn.

Trái ngược với các hệ thống dựa trên thị giác, việc ghi lại chuyển động bằng cảm biến đeo trên người là không phụ thuộc vào môi trường và không nhận biết được sự tắc nghẽn. Nó không yêu cầu thiết lập cơ sở phức tạp và không có ràng buộc về phạm vi chuyển động. Các đặc điểm như vậy làm cho nó phù hợp hơn với các mục đích sử dụng ở cấp độ khách hàng. Do đó, chụp chuyển động bằng cảm biến quán tính ngày càng được chú trọng hơn trong những năm gần đây. Hầu hết các công việc liên quan đều sử dụng Đơn vị đo lường quán tính (IMU) để ghi lại quán tính chuyển động, tiết kiệm, nhẹ, đáng tin cậy và thường được sử dụng trong một số lượng lớn các thiết bị đeo được như đồng hồ, dây đeo tay và kính. Hệ thống mocap quán tính thương mại, Xsens2 [, sử dụng 17 IMU để ước tính độ quay của khớp. Mặc dù chính xác, vị trí dày đặc của các IMU là bất tiện và xâm phạm, ngăn cản người biểu diễn di chuyển tự do. Mặt khác, các yêu cầu về cơ sở vật chất đối với một hệ thống như vậy vẫn nằm ngoài khả năng chấp nhận của các khách hàng bình thường. Công việc của SIP Marcard et al. [2017] chứng minh rằng việc tái tạo lại chuyển động của con người chỉ từ 6 IMU là hoàn toàn khả thi. Tuy nhiên, là một phương pháp dựa trên tối ưu hóa, nó cần phải truy cập vào toàn bộ trình tự và mất nhiều thời gian để xử lý. Công trình hiện đại sau đây, DIP Huang và cộng sự. [2018], đạt được hiệu suất thời gian thực với chất lượng tốt hơn bằng cách tận dụng RNN hai chiều, cũng sử dụng 6 IMU. Tuy nhiên, nó vẫn không thành công ở các tư thế khó và tốc độ khung hình 30 khung hình / giây không đủ để chụp các chuyển động nhanh, rất phổ biến trong các ứng dụng thực tế. Quan trọng hơn, nó chỉ ước tính tư thế cơ thể mà không có chuyển động toàn cầu, vốn cũng quan trọng trong nhiều ứng dụng như VR và AR. Bản thân IMU không có khả năng đo khoảng cách trực tiếp. Một số công trình trước đây Liu et al. 2011; Vlasic và cộng sự. 2007] sử dụng các cảm biến siêu âm bổ sung để đo dịch toàn cầu, đắt tiền và có thể bị tắc. Các giải pháp khả thi khác sử dụng tính năng định vị GPS, cách này không đủ chính xác và chỉ hoạt động khi chụp ngoài trời. Do đó, ước tính siêu thời gian thực của cả tư thế cơ thể và dịch toàn cục từ các cảm biến đeo thưa vẫn còn là một vấn đề còn bỏ ngỏ.

closed 2022-07-09 by NoLoPhe

NoLoPhe, 2022-07-10:
1 Introduction
1.1 Background
Chụp chuyển động của con người (mocap), nhằm mục đích tái tạo lại các chuyển động cơ thể con người 3D, đóng một vai trò quan trọng trong các ứng dụng khác nhau như chơi game, thể thao, y học, VR / AR và sản xuất phim. Cho đến nay, các giải pháp mocap dựa trên tầm nhìn chiếm phần lớn trong chủ đề này. Một danh mục yêu cầu gắn các điểm đánh dấu quang học trên cơ thể người và sử dụng nhiều máy ảnh để theo dõi các điểm đánh dấu để ghi lại chuyển động. Các hệ thống dựa trên điểm đánh dấu như Vicon1 được áp dụng rộng rãi và được coi là đủ chính xác cho việc sử dụng công nghiệp. Tuy nhiên, những cách tiếp cận như vậy yêu cầu cơ sở hạ tầng đắt tiền và các thiết bị xâm nhập, điều này khiến chúng không được mong muốn đối với cách sử dụng ở cấp độ người tiêu dùng. Gần đây, một danh mục khác tập trung vào ước tính tư thế sử dụng một vài máy ảnh RGB hoặc RGB-D Chen et al. Năm 2020; Habibie và cộng sự. Năm 2019; Mehta và cộng sự. Năm 2020; Tome và cộng sự. Năm 2018; Trumble và cộng sự. 2016; Xiang và cộng sự. Năm 2019. Mặc dù các phương pháp này nhẹ hơn nhiều, nhưng chúng rất nhạy cảm với sự xuất hiện của con người vì các đặc điểm phân biệt cần được trích xuất từ hình ảnh. Do đó, các phương pháp này thường hoạt động kém hiệu quả đối môi trường ánh sáng kém. Hơn nữa, tất cả các phương pháp tiếp cận dựa trên thị lực đều bị tắc. Một vấn đề như vậy đôi khi có thể được giải quyết bằng cách đặt các camera dày đặc (điều này càng làm cho hệ thống trở nên nặng nề và tốn kém), nhưng nó thường không thực tế trong một số ứng dụng. Ví dụ, rất khó bố trí máy ảnh trong một phòng sinh hoạt chung có đầy đủ đồ đạc hoặc vật dụng, điều này có thể làm chủ thể bị che khuất từ bất kỳ hướng nào. Một hạn chế khác của phương pháp tiếp cận dựa trên tầm nhìn là những người biểu diễn thường bị hạn chế trong một không gian cố định. Đối với các hoạt động hàng ngày như đi bộ và chạy trong phạm vi lớn, cần phải có camera chuyển động được kiểm soát cẩn thận để ghi lại đủ thông tin chuyển động, điều này rất khó đạt được Xu et al. [2016]. Những nhược điểm này là những sai sót nghiêm trọng đối với nhiều ứng dụng và do đó dẫn đến khả năng sử dụng hạn chế của các giải pháp dựa trên tầm nhìn.
Trái ngược với các hệ thống dựa trên thị giác, việc ghi lại chuyển động bằng cảm biến đeo trên người là không phụ thuộc vào môi trường và không bị tắt nghẽn. Nó không yêu cầu thiết lập cơ sở phức tạp và không có ràng buộc về phạm vi chuyển động. Các đặc điểm như vậy làm cho nó phù hợp hơn với các mục đích sử dụng ở cấp độ khách hàng. Do đó, chụp chuyển động bằng cảm biến quán tính ngày càng được chú trọng hơn trong những năm gần đây. Hầu hết các công việc liên quan đều sử dụng Đơn vị đo lường quán tính (IMU) để ghi lại quán tính chuyển động, tiết kiệm, nhẹ, đáng tin cậy và thường được sử dụng trong một số lượng lớn các thiết bị đeo được như đồng hồ, dây đeo tay và kính. Hệ thống mocap quán tính thương mại, Xsens2, sử dụng 17 IMU để ước tính độ quay của khớp. Mặc dù chính xác, vị trí dày đặc của các IMU là bất tiện, ngăn cản người biểu diễn di chuyển tự do. Mặt khác, các yêu cầu về cơ sở vật chất đối với một hệ thống như vậy vẫn nằm ngoài khả năng chấp nhận của các khách hàng bình thường. Công việc của SIP Marcard et al. [2017] chứng minh rằng việc tái tạo lại chuyển động của con người chỉ từ 6 IMU là hoàn toàn khả thi. Tuy nhiên, là một phương pháp dựa trên tối ưu hóa, nó cần phải truy cập vào toàn bộ trình tự và mất nhiều thời gian để xử lý. Nghiên cứu mới nhất sau đây, DIP Huang và cộng sự. [2018], đạt được hiệu suất thời gian thực với chất lượng tốt hơn bằng cách tận dụng RNN hai chiều, cũng sử dụng 6 IMU. Tuy nhiên, nó vẫn không thành công ở các tư thế khó và tốc độ khung hình 30 khung hình / giây không đủ để chụp các chuyển động nhanh, rất phổ biến trong các ứng dụng thực tế. Quan trọng hơn, nó chỉ ước tính tư thế cơ thể mà không có chuyển động toàn cầu, vốn cũng quan trọng trong nhiều ứng dụng như VR và AR. Bản thân IMU không có khả năng đo khoảng cách trực tiếp. Một số nghiên cứu trước đây Liu et al. [2011]; Vlasic và cộng sự. [2007] sử dụng các cảm biến siêu âm bổ sung để đo dịch toàn cầu, đắt tiền và có thể bị tắc. Các giải pháp khả thi khác sử dụng tính năng định vị GPS, cách này không đủ chính xác và chỉ hoạt động khi chụp ngoài trời. Do đó, ước tính siêu thời gian thực của cả tư thế cơ thể và dịch toàn cục từ các cảm biến đeo vẫn còn là một vấn đề còn bỏ ngỏ.

SlimeVRX, 2022-07-10:
### 3.3.1/. Recurrent Neural Network:

#### 3.3.1.1/. Definition:

RNN is a powerful class of artificial network typically uses sequential or time series data, based on the work of Rumelhart’s work for modelling time series [16]. A notable distinguishing feature of RNN is that it has “memory” that stores the states of previous inputs to influence the next output of the sequence, whereas an ordinary deep neural network considers inputs and outputs to be independent of each other. In order to fully explain the mechanism of RNN, a working knowledge of a typical feed-forward neural network is necessary. Figure 1 illustrates how information moves in RNN and feed-forward neural networks.

[image]

Figure 1. Comparison of RNN and Feed-forward Neural Networks

We refer to this example of the difference in information flow between RNN and feed-forward neural network for a detailed explanation [18].

#### 3.3.1.2/. Types:

Figure 2 indicates different types of RNN

[image]

Figure 2. Types of RNN

#### 3.3.1.3/. Limitations:

There are 2 major obstacles to the performance of RNN: exploding gradient and vanishing gradient.

##### 3.3.1.3.1/. Exploding gradients:

Exploding gradients happen when large error gradients accumulate and result in very large updates to neural network model weights during training. This issue typically can be solved by truncating or squashing the gradients

##### 3.3.1.3.2/. Vanishing gradients:

Vanishing gradients are when the values of gradients are too small and the model either stops or takes a very long time to learn. Unlike exploding gradients which can be easily solved by truncating or squashing the gradients, this was a major problem in the 1990s. Later, it was properly addressed with the introduction of LSTM.

### 3.3.2/. LSTM:

#### 3.3.2.1/. Definition:

LSTM is “a specific recurrent neural network (RNN) architecture that was designed to model temporal sequences and their long-range dependencies more accurately than conventional RNNs” [19] and it was first brought up by Hochreiter and Schmidhuber to address the problem of vanishing gradient in RNN [20]. Similar to RNN, LSTM also has a hidden state which acts as the short-term memory of the network. Moreover, LSTM adds a cell state which functions as the long-term memory of LSTM. Information added to the cell state or removed from it are controlled by gates. There are three gates in the structure of LSTM: forget gate, input gate and output gate. Figure 3 illustrates the components of a LSTM cell.

[image]

Figure 3. The components of a LSTM cell

#### 3.3.2.2/. Activation functions in LSTM:

There are 2 activation functions in used in a LSTM cell: the Sigmoid function and the Tanh activation function

#### 3.3.2.2.1/. Sigmoid function:

Figure 4 shows the curve and equation of the Sigmoid function.

[image]

Figure 4. The Sigmoid function [21]

The Sigmoid function is used as the gating function for three gates (Forget, Input and Output) in LSTM because its output is a value between 0 and 1. Only such values can dictate the decisions required in a gate (From forgetting everything to remember everything).

The properties of the sigmoid function:
- It is differentiable.
- It is monotonic unlike its derivative.
- It can cause the neural networks to stuck.

#### 3.3.2.2.2/. Tanh activation function:

Figure 5 shows the curve of the Tanh activation function compare to the Sigmoid function.

[image]

Figure 5. The Tanh activation function [21]

The intuition of using tanh function in LSTM can be explained as of follows. The Sigmoid function is used as the gating function because it can let either no information or complete flow of information pass through the gates. In order to bypass the vanishing gradient issue, a function whose second derivative can sustain for a long range before going to zero. And the tanh function meets all the criteria above.

#### 3.3.2.2/. The mechanism of LSTM:

Notation: For clarification purposes, we first explain the notations used in this section’s equations
- σ: represents the sigmoid function.
- xt: the input to the current timestamp.
- Ux: the weight matrix associated with inputs for the respective gates.
- Ht: the hidden state of the current timestamp.
- Ht-1: the hidden state of the previous timestamp.
- Wx: the weight matrix associated with the hidden state for the respective gates.
- Nt: vector of new candidate values
- bx: biases for the respective gates
- Ct: cell state at the current timestamp
- Ct-1: cell state at the previous timestamp

A LSTM cell takes the previous cell state and hidden state (Ht-1 and Ct-1) as inputs and generate a new cell state and hidden state as outputs. Firstly, LSTM decides whether to keep or throw away information from the previous timestamp (ct-1). This decision is made by the forget gate. We have this equation for the forget gate:

ft = σ (xt * Uf + Ht-1 * Wf + bf) (2.1)

The next step is deciding what new information we are going to add to the cell state. Firstly, the input gate decides which values will be updated. Secondly, a tanh layer generates a vector of new candidate values, Nt. These are the equations used in both parts of the second step:

it = σ (xt * Ui + Ht-1 * Wi + bi) (2.2)

Nt = tanh (xt * Uc + Ht-1 * Wc + bc) (2.3)

Next, after obtaining the calculations from both forget gate and input gate, LSTM updates the current cell state using the following equation:

Ct = ft * Ct-1 + it * Nt (2.4)

Finally, LSTM calculates the current hidden state with the following formulas:

ot = σ (xt * Uo + Ht-1 * Wo + bo) (2.5)

Ht = ot * tanh(Ct) (2.6)

SlimeVRX, 2022-07-10:
### 3.3.3/. Transpose Neural Network:

[image]

Fig. 2. Overview of our pipeline. We divide our task into 2 subtasks: multi-stage pose estimation and fusion-based global translation estimation. The pose estimation subtask is formulated into 3 stages, where: 1) Pose-S1 estimates 5 leaf joint positions relative to the root from IMU signals, and its output is concatenated to the inertial measurements as 𝒙 (1) ; 2) Pose-S2 estimates all joint positions from 𝒙 (1) , and the output is again concatenated to the inertial measurements as 𝒙 (2) ; 3) Pose-S3 regresses the joint rotations 𝜽 from 𝒙 (2) . The translation estimation subtask is addressed by two parallel branches, where:
1) Trans-B1 estimates foot-ground contact probabilities from 𝒙 (1) and calculates the root velocity 𝒗 𝑓 using forward kinematics assuming that the foot with a higher probability is still; 
2) Trans-B2 estimates the local velocity of the root joint 𝒗¯𝑒 in its own coordinate frame from 𝒙 (2) , which is then transformed to the world space as 𝒗𝑒 according to the root rotation measured by the sensor mounted on the pelvis. Such two estimated velocities are then fused to form the final global translation 𝒗 according to the foot-ground contact probabilities.

### 3.3.3.1/. System Input: Đầu vào hệ thống

Chúng tôi lấy số đo vòng quay và gia tốc của mỗi IMU làm đầu vào tổng thể của hệ thống. Chúng tôi sắp xếp các phép đo này vào cùng một hệ quy chiếu và chuẩn hóa chúng để thu được vectơ đầu vào nối liền là 𝒙 (0) = [Rroot, · · ·, Rrarm, Rroot, · · ·, Rrarm] ∈ R 72 trong đó 𝒂 ∈ R 3 là gia tốc và 𝑹 ∈ R 3 × 3 là ma trận quay. Chúng tôi sử dụng 𝒙 (0) (𝑡) để chỉ các phép đo của khung thứ 𝑡 và chỉ số trên (0) có nghĩa là nó là đầu vào tổng thể. Vui lòng tham khảo Phụ lục A để biết thêm chi tiết về tiền xử lý cảm biến.

### 3.3.3.2/. Multi-stage Pose Estimation‌:

In this section, we introduce our multi-stage method to estimate body poses, i.e. the rotation of each joint, from sparse IMU measurements. This task is very difficult due to the ambiguity of the mapping from extremely sparse inertial data to full body joint angles. The key to address this challenging task is the use of 1) the prior knowledge on human poses and 2) the temporal information. Because of the diversity of human motions and the anatomic restrictions, the pose prior is hard to model or learn. While the previous work, DIP [Huang et al 2018], proposes to adopt a bidirectional recurrent neural network (biRNN) to learn the direct mapping from IMU measurements to joint rotations, we demonstrate that using joint positions as an intermediate representation is significantly helpful for the model to learn the complex motion prior. More specifically, we first predict joint positions as an intermediate task in Pose-S1 (short for pose estimation stage 1) and Pose-S2 (Section 3.2.1), and then solve for the joint angles in Pose-S3 (Section 3.2.2). Further, to fully utilize the temporal information, we adopt biRNNs to leverage the past and future frames. As shown in Section 5.2, our approach achieves superior capture quality and runtime performance.

#### 3.3.3.2.1/. Pose-S1&S2: estimating joint positions:

In Pose-S1&S2, we estimate joint positions from IMU measurements. We explicitly separate this task into two different stages to take the hierarchy of the human kinematic tree into consideration. Specifically, we divide the human joints into two sets: the leaf joints and the non-leaf joints. The leaf joints are mounted with sensors and have direct measurements. Usually, they have larger movements. The inner non-leaf joints are without any direct inertial data and have relatively smaller motions. Due to the correlation between body joints, the information of the leaf joints is helpful for predicting the coordinates of inner ones. We first regress the positions of the leaf joints in Pose-S1, and then localize the other joints in Pose-S2.

The input to Pose-S1 is the inertial measurement vector 𝒙 (0) 𝑡 , and the output is the root-relative positions of the five leaf joints 𝒑leaf (𝑡) = [𝒑lleg (𝑡), · · · , 𝒑rarm(𝑡)] ∈ R15. We use a standard biRNN [Schuster and Paliwal 1997] with Long Short-Term Memory (LSTM) cells [Hochreiter and Schmidhuber 1997] to learn the mapping from IMU measurements to leaf joint positions. Due to the very sparse measurements, it is common that the IMUs show identical signals while the subject is performing different animations. For example, when sitting still or standing still, the acceleration and orientation measurements of all the IMUs are almost the same. In such cases, the temporal information is the key to resolve this ambiguity, thus using RNNs is a natural choice. We choose to use biRNNs instead of vanilla RNNs because the future information is also of great help in this task, as stated in DIP. The loss function used to train the network is defined as:

LS1 = ∥𝒑leaf (𝑡) − 𝒑GTleaf (𝑡) ∥22, (1)

where superscript GT denotes the ground truth.

The input of Pose-S2 is the concatenation of Pose-S1’s output and the inertial measurements: 𝒙 (1) 𝑡 = 𝒑leaf 𝑡 , 𝒙 (0) 𝑡 R87. Based on the input information, Pose-S2 outputs the root-relative coordinates of all joints: 𝒑all(𝑡 ) = [𝒑 𝑗 (𝑡 ) | 𝑗 = 1, 2, · · · , 𝐽 − 1] ∈ R3( 𝐽 −1) , where 𝐽 is the number of joints in the human kinematic tree. Similar to Pose-S1, we use a biRNN for Pose-S2 with L2 loss:

LS2 = 𝒑all(𝑡 ) − 𝒑GT(𝑡) 2. (2)

#### 3.3.3.2.2/. Pose-S3: estimating joint rotations:

[... cut ...] Zhou et al 2018]. We should note that the IMU measurements are fed into the networks as rotation matrices, while the only use of the 6D representation is in the output of Pose-S3. The loss function is defined as:

LS3 = 𝑹 (6D) (𝑡 ) − 𝑹GT,(6D) (𝑡 ) 2. (3)

The rotation of the root joint 𝑹root is directly measured by the sensor placed on the pelvis. Combining all these rotations, we convert them to the rotation matrix formulation and obtain the full body as 𝜽 = [𝑹𝑗 (𝑡)|𝑗 = 0, 1, · · · , 𝐽] ∈ R9𝐽
.

SlimeVRX, 2022-07-10:
### 3.3.3.3/. Fusion-based Global Translation Estimation:

In this section, we explain our method to estimate global translations from the IMU measurements and the estimated body poses. This task is even more challenging due to the lack of direct distance measurements, and the acceleration measurements are too noisy to be used directly [Marcard et al 2017]. Previous works address this task by introducing additional vision inputs [Andrews et al 2016]; [Henschel et al 2020]; [Malleson et al 2019, 2017] or distance measurements [Liu et al 2011]; [Vlasic et al 2007], which increase the complexity of the system. While the work of SIP [Marcard et al 2017] estimates global translations from IMUs only, it has to run in an offline manner. To our best knowledge, we are the first method that addresses the task of real-time prediction of global translations from sparse IMUs. To do so, we need to estimate the per-frame velocity of the root joint, i.e. the translation between the current frame and the previous frame. We propose a fusion-based approach that comprises two parallel branches. In Trans-B1 (short for translation estimation branch 1) (Section 3.3.1), we infer the root velocity based on the sequential pose estimation, in combination with the prediction of foot-ground contact probabilities. In Trans-B2 (Section 3.3.2), we use an RNN to predict the root velocity from joint positions and inertial measurements. These two branches run in a parallel style, and the final estimation is a fusion of the two branches based on the foot state (Section 3.3.3). The intuition behind is that by assuming the foot on the ground is not moving, the per-frame velocity can be deduced from the motion of the subject. However, this estimation is not totally accurate and inevitably fails when both feet are not on the ground, e.g., jumping or running. Therefore, a neural network is used here for complementary estimation. We demonstrate in Section [5.3] that such a hybrid approach gives more accurate results than using either branch alone.

#### 3.3.3.3.1/. Trans-B1: supporting-foot-based velocity estimation

In Trans-B1, we estimate the velocity of the root joint based on the estimation of body poses. First, we use a biRNN to estimate the foot-ground contact probability for each foot, i.e. the likelihood that the foot is on the ground, formulated as

SlimeVRX, 2022-07-12:
### 2.1\. Tổng quan về cảm biến quán tính IMU:
#### 2.1.1\. Cảm biến quán tính IMU là gì?:

Inertial Measurement Unit = Đơn vị đo lường quán tính, có chức năng cung cấp thông tin về tốc độ góc và góc nghiêng của hệ thống. Dựa trên nguyên lý hoạt động của 2 cảm biến accelemeter (gia tốc kế) và gyroscope ( con quay hồi chuyển ). Tuy nhiên khái niệm này được sử dụng khá lỏng lẻo: do vậy IMU có thể hiểu là đơn vị đo lường quán tính của hệ thống, cung cấp các giá trị cảm biến sau khi đã xử lý giúp cho hệ thống hoạt động tốt nhất.

IMU  cơ bản sẽ bao gồm 6-DOF (6 Degrees Of Freedom) tức là 6 trục độc lập (3 của accel và 3 của gyro). Tuy vậy đôi khi như thế vẫn là không đủ, những dự án phức tạp như là điều khiển máy bay hoặc robot có thể sẽ cần đến 9-DOF (thêm một cảm biến từ trường 3 trục – magnetometer – hoạt động gần giống một la bàn để định hướng), hoặc 10-DOF (thêm một áp kế – barometer – dùng để đo độ cao) hoặc thậm chí 11-DOF (thêm module GPS để xác định vị trí).

#### 2.1.2\. Ứng dụng của IMU:

Các cảm biến IMU được sử dụng trong các ứng dụng như: robot tự cân bằng, quadcopter, điện thoại thông minh hay đề tài chúng em đang nghiên cứu là ước tính cơ thể người…. Cảm biến IMU giúp chúng ta có được vị trí của vật thể gắn với cảm biến trong không gian ba. Chúng được sử dụng để phát hiện hướng của điện thoại thông minh hoặc trong các tiện ích như Fitbit, sử dụng cảm biến IMU để theo dõi chuyển động.
Module cảm biến IMU có thể được tích hợp gồm hai hoặc nhiều cảm biến như:
•	Cảm biến gia tốc (Accelemeter)
•	Cảm biến góc quay (Gyroscope)
•	Cảm biến từ trường (Magnetometer)

#### 2.1.3\. Nguyên lý hoạt động của từng loại cảm biến:

##### 2.1.3.1\ Cảm biến gia tốc kế Accelemeter:

Khi bạn đặt một con chip IMU để im không chuyển động, giá trị trả về gyro = [0.0, 0.0, 0.0] do không có bất cứ chuyển động quay nào cả. Gyro chỉ đo tốc độ quay chứ không đo trực tiếp góc quay, nên khi bạn quay module một góc nào đó rồi dừng, giá trị của gyro sẽ tăng lên rồi hạ xuống về 0.

##### 2.1.3.2\ Cảm biến con quay hồi chuyển Gyroscope:

Nguyên tắc để đo góc dùng gia tốc kế là phân tích sự tác dụng của trọng lực lên các trục ta có thể dùng nó để tính các góc lệch roll và pitch.

##### 2.1.3.3\ Cảm biến từ trường Magnetometer:

[Hình]

Cảm biến từ trường đa phần hoạt động dựa trên hiệu ứng Hall. Ta cấp nguồn cho dòng eletron chạy qua mạch, có một tấm dẫn điện như trên hình.

[Hình]

Khi xung quanh nó có từ trường, lực Lorent sẽ làm dòng eletron chạy trong mạch nó di chuyển lệch đi về 2 phía. Nếu ta đo điện áp thì sẽ biết được độ lớn của nguồn từ trường này.

Nếu chọn trục z là trục vuông góc với mặt đất thì góc yaw là góc khi xoay trục z. Nó thường được ứng dụng để đo góc yaw bằng công thức đơn giản. Trên thực tế ta có thể tính toán góc yaw bằng gyroscope, nhưng nếu ta kết hợp thêm cảm biến từ trường nó sẽ cho kết quả chính xác hơn, lấp đi nhược điểm khi dùng gyroscope.

#### 2.1.4\. Đặc tính của một số loại cảm biến:

##### 2.1.4.1\ Cảm biến gia tốc kế Accelemeter:

Vấn đề thường gặp phải nhất của gyro là drift = độ trôi, nó thay đổi chậm theo thời gian. Nguyên nhân bởi vì do các tác động cơ khí, rung động tác động lên gyro, sau một thời gian sử dụng thì giá trị trôi này tích lũy lên đáng kể, làm giá trị đo góc không còn chính xác.

Dù vậy, điểm mạnh của gyro là ít bị nhiễu hơn accelemeter, nghĩa là giá trị tức thời của nó đáng tin cậy.

##### 2.1.4.2\ Cảm biến con quay hồi chuyển Gyroscope:

Accel luôn có offset trên mỗi trục làm cho giá trị đo được thường lệch đi so với thực tế một chút. Ngoài ra, giá trị đó được theo accel thường nhạy với rung động cơ khí nhỏ khiến cho giá trị tức thời của nó không đáng tin cậy, do đó chúng ta chỉ sử dụng giá trị trung bình của acc thì nó mới có hữu hiệu, vì nếu để lâu dài thì accel không bị trôi như gyro.

#### 2.1.5\. Các giải thuật tính toán IMU

### 2.2\. Cảm biến Hi229

#### 2.2.1\. Mô tả chung

Hi229 do HiPNUC sản xuất là Hệ thống trong gói (SiP) tích hợp gia tốc kế ba trục, con quay hồi chuyển ba trục, từ kế ba trục và vi điều khiển 32-bit ARM® Cortex ™ -M4 chạy chương trình cơ sở kết hợp cảm biến của HiPNUC. Phần sụn cung cấp các thuật toán xử lý tín hiệu phức tạp để xử lý dữ liệu cảm biến và cung cấp định hướng 3D thời gian thực chính xác, tiêu đề, gia tốc đã hiệu chỉnh và vận tốc góc đã hiệu chỉnh, cũng như dữ liệu cảm biến thô đã được hiệu chỉnh. Hi229 có một số đặc tính chống nhiễu từ trường nhất định trong nhà và vẫn có thể hoạt động bình thường trong môi trường nhiễu từ trường cường độ nhất định.

#### 2.2.2\. Các tính năng chính
#### 2.2.3\. Các tính năng chính của cảm biến tích hợp
Ứng dụng tiêu biểu
Kết nối Hi229

Đặc điểm hoạt động
Độ chính xác góc

SlimeVRX, 2022-07-12:
Chúng tôi lấy số đo vòng quay và gia tốc của mỗi IMU làm đầu vào tổng thể của hệ thống. Chúng tôi sắp xếp các phép đo này vào cùng một hệ quy chiếu và chuẩn hóa chúng để thu được vectơ đầu vào nối liền là 𝒙 (0) = [Rroot, · · ·, Rrarm, Rroot, · · ·, Rrarm] ∈ R 72 trong đó 𝒂 ∈ R 3 là gia tốc và 𝑹 ∈ R 3 × 3 là ma trận quay. Chúng tôi sử dụng 𝒙 (0) (𝑡) để chỉ các phép đo của khung thứ 𝑡 và chỉ số trên (0) có nghĩa là nó là đầu vào tổng thể. Vui lòng tham khảo Phụ lục A để biết thêm chi tiết về tiền xử lý cảm biến.

SlimeVRX, 2022-07-13:
[image]

https://hocday.com/li-cam-oan-v3.html#DANH_M%E1%BB%A4C_C%C3%81C_T%E1%BB%AA_VI%E1%BA%BET_T%E1%BA%AET_V%C3%80_K%C3%9D_HI%E1%BB%86U

SlimeVRX, 2022-07-13:
Gia tốc là một cảm biến quán tính được sử dụng phổ biến để đo gia tốc tuyến tính

[image]

Gia tốc kế đo gia tốc cảm nhận theo ba trục, vì vậy khi đặt gia tốc kế nằm ngang và đứng yên trên trái đất, số đọc là 0,0,1G. Khi gia tốc kế nhận được gia tốc 1G theo chiều dương của trục X, số đọc là 1,0,1G

Trọng lực hướng xuống theo phương thẳng đứng, nhưng mặt trước được đặt theo phương ngang, và độ là 0,0,1G, tại sao không phải 0,0, -1G. Điều này là do gia tốc kế cảm nhận được lực tác động lên anh ta từ bên ngoài (đặt gia tốc kế như mô hình lò xo). chứ không phải là một trường hấp dẫn.

[image]

Mô hình gia tốc kế

SlimeVRX, 2022-07-13:
# Cảm biến trường địa từ

Cảm biến trường địa từ đo cường độ từ trường theo ba trục (từ trường địa từ và các từ trường gây nhiễu khác gần đó chẳng hạn như nam châm, vật kim loại,...)

[image]

Cái gọi là cảm biến 9 trục, hoặc MARG (từ tính, tốc độ góc), có thể được gọi là AHRS (hệ quy chiếu tiêu đề thái độ) với một thuật toán phù hợp nếu nó được trang bị một máy vi tính chip đơn.

Cảm biến từ trường ba trục cảm nhận từ trường xung quanh: cảm biến từ lý tưởng có thể được mô tả như mô hình toán học sau:

[image]

[image] Tổng của từ trường trái đất và từ trường của môi trường xung quanh
[image]
 thiên vị cảm biến
[image]
 cảm biến tiếng ồn

SlimeVRX, 2022-07-13:
Từ trường của trái đất

Sự phân bố hình học của từ trường trái đất rất không đều. Nói chung, các đường sức từ bắt đầu từ cực nam và quay trở lại cực bắc của từ trường:

[image]

Đối với bán cầu Bắc, hướng của địa từ nghiêng xuống và đối với bán cầu Nam, hướng của địa từ nghiêng lên trên.

[image]

[IMG] Một trường địa từ điển hình ở bán cầu bắc

https://zhuanlan.zhihu.com/p/56073859

SlimeVRX, 2022-07-13:
Đơn vị đo lường quán tính - IMU thường được đề cập đến như một đơn vị kết hợp bao gồm cảm biến gia tốc kế 3-axis và cảm biến con quay hồi chuyển 3-axis

https://www.ceva-dsp.com/ourblog/what-is-an-imu-sensor/#:~:text=An%20IMU%20is%20a%20specific,considered%20a%209%2Daxis%20IMU.

SlimeVRX, 2022-07-13:
## 2.4. Các giải thuật tính toán IMU:

Roll, Pitch, Yaw, Quaternion

SlimeVRX, 2022-07-13:
Hi229

SlimeVRX, 2022-07-13:
Transpose Model

SlimeVRX, 2022-07-13:
LSTM Model