SlimeVRX/SlimeVRX #36: [Bảng vẽ]

SlimeVRX, 2022-07-12:
[banveluudo-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9094596/banveluudo-Model.pdf)

[banvemachdien-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9094597/banvemachdien-Model.pdf)

NoLoPhe, 2022-07-16:
[banvemachdien-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9124846/banvemachdien-Model.pdf)

NoLoPhe, 2022-07-16:
[Uploading sodokhoiketnoicambien.pdf…]()

NoLoPhe, 2022-07-16:
[sodokhoiketnoicambien.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9125491/sodokhoiketnoicambien.pdf)

NoLoPhe, 2022-07-17:
[banveketnoicambien-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127847/banveketnoicambien-Model.pdf)
[banveluudothuattoan-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127848/banveluudothuattoan-Model.pdf)
[banvevitricambien-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127849/banvevitricambien-Model.pdf)

NoLoPhe, 2022-07-17:
[banveketnoicambien-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9128743/banveketnoicambien-Model.pdf)
[banveluudothuattoan-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9128744/banveluudothuattoan-Model.pdf)
[banvevitricambien-Model.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9128745/banvevitricambien-Model.pdf)