SlimeVRX/SlimeVRX #38: [doc]

SlimeVRX, 2022-07-12:


SlimeVRX, 2022-07-12:
[FILE_20220708_174937_thuyetminhdatn (1).pdf](https://github.com/SlimeVRX/SlimeVRX/files/9097147/FILE_20220708_174937_thuyetminhdatn.1.pdf)

SlimeVRX, 2022-07-13:
[MAIN_ThuyetMinh.docx](https://github.com/SlimeVRX/SlimeVRX/files/9097862/MAIN_ThuyetMinh.docx)

SlimeVRX, 2022-07-13:
[3.2.1.docx](https://github.com/SlimeVRX/SlimeVRX/files/9098102/3.2.1.docx)

SlimeVRX, 2022-07-13:
[thuyetminh_supermain.docx](https://github.com/SlimeVRX/SlimeVRX/files/9098604/thuyetminh_supermain.docx)

SlimeVRX, 2022-07-13:
[thuyetminh_supermain_1447.docx](https://github.com/SlimeVRX/SlimeVRX/files/9100046/thuyetminh_supermain_1447.docx)

NoLoPhe, 2022-07-13:
[
[Uploading thuyetminh_supermain_1752.docx…]()
](url)

NoLoPhe, 2022-07-13:
[Uploading thuyetminh_supermain_1752.docx…]()

NoLoPhe, 2022-07-13:
[thuyetminh_supermain_1752.docx](https://github.com/SlimeVRX/SlimeVRX/files/9101383/thuyetminh_supermain_1752.docx)

NoLoPhe, 2022-07-13:
[thuyetminh_supermain_1752.docx](https://github.com/SlimeVRX/SlimeVRX/files/9101650/thuyetminh_supermain_1752.docx)

NoLoPhe, 2022-07-13:
[mathtype.docx](https://github.com/SlimeVRX/SlimeVRX/files/9101884/mathtype.docx)

NoLoPhe, 2022-07-13:
[thuyetminh_supermain_2215.docx](https://github.com/SlimeVRX/SlimeVRX/files/9103594/thuyetminh_supermain_2215.docx)

SlimeVRX, 2022-07-13:
[thuyetminh_supermain_140613.docx](https://github.com/SlimeVRX/SlimeVRX/files/9106676/thuyetminh_supermain_140613.docx)

SlimeVRX, 2022-07-14:
[thuyetminh_supermain_140812.docx](https://github.com/SlimeVRX/SlimeVRX/files/9107065/thuyetminh_supermain_140812.docx)

NoLoPhe, 2022-07-14:
[thuyetminh_supermain_140812 (Repaired).docx](https://github.com/SlimeVRX/SlimeVRX/files/9108037/thuyetminh_supermain_140812.Repaired.docx)

NoLoPhe, 2022-07-14:
[thuyetminh_supermain_1005.docx](https://github.com/SlimeVRX/SlimeVRX/files/9108052/thuyetminh_supermain_1005.docx)

SlimeVRX, 2022-07-14:
[thuyetminh_supermain_141305.docx](https://github.com/SlimeVRX/SlimeVRX/files/9109056/thuyetminh_supermain_141305.docx)

SlimeVRX, 2022-07-14:
Gửi thầy:

[thuyetminh_supermain_141305.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9109185/thuyetminh_supermain_141305.pdf)

SlimeVRX, 2022-07-14:
[thuyetminh_supermain_141352.docx](https://github.com/SlimeVRX/SlimeVRX/files/9109288/thuyetminh_supermain_141352.docx)

NoLoPhe, 2022-07-14:
[thuyetminh_supermain_1724.docx](https://github.com/SlimeVRX/SlimeVRX/files/9110998/thuyetminh_supermain_1724.docx)

SlimeVRX, 2022-07-15:
[Edit_thuyetminh_supermain_150838.docx](https://github.com/SlimeVRX/SlimeVRX/files/9116995/Edit_thuyetminh_supermain_150838.docx)

NoLoPhe, 2022-07-15:
[Edit_thuyetminh_supermain_150915.docx](https://github.com/SlimeVRX/SlimeVRX/files/9117135/Edit_thuyetminh_supermain_150915.docx)

NoLoPhe, 2022-07-15:
[Edit_thuyetminh_supermain_151136.docx](https://github.com/SlimeVRX/SlimeVRX/files/9117836/Edit_thuyetminh_supermain_151136.docx)

SlimeVRX, 2022-07-15:
[Edit_thuyetminh_supermain_151450.docx](https://github.com/SlimeVRX/SlimeVRX/files/9118942/Edit_thuyetminh_supermain_151450.docx)

NoLoPhe, 2022-07-15:
[Edit_thuyetminh_supermain_151529.docx](https://github.com/SlimeVRX/SlimeVRX/files/9119199/Edit_thuyetminh_supermain_151529.docx)

SlimeVRX, 2022-07-15:
[Edit_thuyetminh_supermain_151805.docx](https://github.com/SlimeVRX/SlimeVRX/files/9120287/Edit_thuyetminh_supermain_151805.docx)

SlimeVRX, 2022-07-15:
[Edit_thuyetminh_supermain_151955.docx](https://github.com/SlimeVRX/SlimeVRX/files/9120918/Edit_thuyetminh_supermain_151955.docx)

NoLoPhe, 2022-07-15:
[Edit_thuyetminh_supermain_152138.docx](https://github.com/SlimeVRX/SlimeVRX/files/9121723/Edit_thuyetminh_supermain_152138.docx)

SlimeVRX, 2022-07-16:
[Edit_thuyetminh_supermain_160943.docx](https://github.com/SlimeVRX/SlimeVRX/files/9125494/Edit_thuyetminh_supermain_160943.docx)

NoLoPhe, 2022-07-16:
[Edit_thuyetminh_supermain_162212 (1).docx](https://github.com/SlimeVRX/SlimeVRX/files/9126073/Edit_thuyetminh_supermain_162212.1.docx)

SlimeVRX, 2022-07-17:
[Edit_thuyetminh_supermain_170936.docx](https://github.com/SlimeVRX/SlimeVRX/files/9127159/Edit_thuyetminh_supermain_170936.docx)

[thuyetminh_171248.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127160/thuyetminh_171248.pdf)

[banve_vitricb.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127161/banve_vitricb.pdf)

[banve_sodokhoiketnoicambien.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127162/banve_sodokhoiketnoicambien.pdf)

[banve_luudothuattoan.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127163/banve_luudothuattoan.pdf)

SlimeVRX, 2022-07-17:
[thuyetminh_supermain_171918.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127599/thuyetminh_supermain_171918.pdf)

SlimeVRX, 2022-07-17:
Gửi In:

[Ghep_thuyetminh_supermain_170936.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127957/Ghep_thuyetminh_supermain_170936.pdf)

[Ghep_2.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9127958/Ghep_2.pdf)

SlimeVRX, 2022-07-17:
[bia.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9128769/bia.pdf)