SlimeVRX/SlimeVRX #45: [paper] Slide đồ án

SlimeVRX, 2022-07-17:
1. Giới thiệu

VRchat là một game mà người chơi có tương tác với nhau bằng hành 
- giao tiếp bằng ngôn ngữ, hình ảnh
- giao tiếp bằng tương tác vật lý với người hoặc vật
- tương tác, điều khiển sự vật thực tế từ môi trường ảo.

Bộ thực tế ảo thông thường chỉ hỗ trợ theo dõi đầu (HMD) và 2 tay (controller)
Một hệ thống hỗ trợ full body tracking trong game VRChat là một thách thức
- SlimeVR sử dụng 5 cảm biến quán tính + 1 HMD + 2 controller
- Mediapipe sử dụng 1 camera RGB
- Kinenct sử dụng 1 camera chiều sâu
Tóm lại: yêu cầu một hệ thống hỗ trợ nhận dạng khung xương 3D trong thời gian thực
Tình cờ tìm được Transpose paper, một nghiên cứu sử dụng 6 cảm biến quán tính (IMU) để ước tính cơ thể và chuyển động trong không gian 3D sử 6 cảm biến quán tính. Ước tính chuyển động trong không gian 3D là một điểm nhấn trong bài báo này, phương pháp là ước tính chân nào chạm đất trước (thuật toán động học kine Dymatic)

Thực hiện: Tái tạo lại kết quả của bài báo transpose:
- Sử dụng cảm biến IMU giá thấp thay thế cho cảm biến IMU chuyên dụng của Noitom
-

SlimeVRX, 2022-07-18:
[image]

SlimeVRX, 2022-07-18:
Thiếu sót, mọi người không hiểu công việc của mình hoạt động như thế nào

Phương pháp làm dữ liệu dựa trên máy học

Trainning dựa trên chuỗi hành động AMASS

Học LSTM

Có 5 khối LSTM

[image]

SlimeVRX, 2022-07-18:
So sanh người độ cao khác nhau ảnh hưởng như thế nào đến việc predict

SlimeVRX, 2022-07-18:
Các khối trong Transpose model
Viết lại LSTM, các nút nhớ, quên

SlimeVRX, 2022-07-18:
[image]

SlimeVRX, 2022-07-18:
[image]

SlimeVRX, 2022-07-18:
[image]

SlimeVRX, 2022-07-18:
Phải thể hiện rõ vừa ước tính tư thế, vừa ước tính di chuyển trong không gian 3D

SlimeVRX, 2022-07-19:
[image]

Mục tiêu: Thế giới công nghệ luôn thay đổi và phát triển. Những tiến bộ trong công nghệ giúp kết nối giữa con người và con người trở nên dễ dàng hơn, bắt đầu từ gửi mẫu tin nhắn bằng mã Morse đến phát minh ra điện thoại. Hiện nay, nhờ sự phát triển của Metaverse, trong thế giới ảo con người giao tiếp với nhau qua hành động của cơ thể. Ước tính cơ thể người từ cảm biến quán tính phù hợp với xu hướng này. Các ứng dụng AR, VR, game Play to Move yêu cầu hoạt động mọi lúc mọi nơi khiến ước tính cơ thể người từ thị giác máy tính không còn phù hợp.

SlimeVRX, 2022-07-19:
Tôi muốn chứng minh mạng neural hoạt động với nhiều kích thước cơ thể khác nhau, chiều cao khác nhau 1m6, 1m7, 1m77

SlimeVRX, 2022-07-19:
Tôi muốn chứng minh, tối ưu tốc độ, độ chính giảm nhưng nằm trong ngưỡng chấp nhận được

SlimeVRX, 2022-07-19:
Cách tính độ delay?

SlimeVRX, 2022-07-19:
1 slide so sánh

SlimeVRX, 2022-07-19:
[image]

Bộ dữ liệu SMPL

SlimeVRX, 2022-07-19:
[image]

SlimeVRX, 2022-07-19:
[image]

[image]

SlimeVRX, 2022-07-19:
Lỗi của Transpose model

Lỗi của Transpose Light model

SlimeVRX, 2022-07-19:
[image]

SlimeVRX, 2022-07-19:
[image]

SlimeVRX, 2022-07-19:
Định tính, định lượng, live demo

SlimeVRX, 2022-07-21:
[slide_tn.pptx](https://github.com/SlimeVRX/SlimeVRX/files/9156859/slide_tn.pptx)

SlimeVRX, 2022-07-21:
[BaocaoTN.pptx](https://github.com/SlimeVRX/SlimeVRX/files/9160148/BaocaoTN.pptx)

SlimeVRX, 2022-07-21:
[image]

SlimeVRX, 2022-07-21:
https://drive.google.com/u/0/uc?id=0B1hPpa4R2wZMM09VYlpzcnNqNzA&export=download

NoLoPhe, 2022-07-21:
[BaocaoTN1.pptx](https://github.com/SlimeVRX/SlimeVRX/files/9162349/BaocaoTN1.pptx)

SlimeVRX, 2022-07-21:
```
317572/ 322999 frame (98.32%) error: 5427 frame

tensor([[13.9808,  6.8008],
        [ 7.1220,  3.6887],
        [ 4.7241,  2.7607],
        [ 5.4321,  3.1413],
```

NoLoPhe, 2022-07-21:
[yyyy - Copy.pptx](https://github.com/SlimeVRX/SlimeVRX/files/9162994/yyyy.-.Copy.pptx)

SlimeVRX, 2022-07-21:
[BaocaoTN_best.pdf](https://github.com/SlimeVRX/SlimeVRX/files/9163345/BaocaoTN_best.pdf)

SlimeVRX, 2022-08-16:
Video: https://drive.google.com/drive/folders/1ZRSDt-3VDnrgFhXa0-fwevu886yFKO73?usp=sharing