SlimeVRX/SlimeVRX #46: [code] Transpose

NoLoPhe, 2022-07-21:


MichaelLiLee, 2022-11-06:
Hi SlimeVRX Team,
I try to reimplement TransPose but have trouble with,
then i find this repository, and saw your team have successfully reimplement(i found implement video in this repository, it seems work great).

can you provide assistance to help me reimplement this Project?

i use six Hi221 sensor from HiPNUC，
i read the code from code_1.zip file and can't found normalize_and_concat_np function from src.utils and TransPoseNet_np from src.net_ONNX
can you provide the soure code for it ?

Thanks!
Michael

SlimeVRX, 2022-11-07:
Hi @MichaelLiLee 

Please add Wechat me, I'm too busy right now, I'll help you when I'm free.

SlimeVRX, 2022-11-07:
[image]

gsygsy96, 2023-01-10:
Hi @SlimeVRX ,  would you mind help me? My wechat is Guan-Shyan

SlimeVRX, 2023-01-10:
Hi @gsygsy96 

Please add me!

[image]

clxee, 2025-12-16:
I was also using my imu to reproduce the paper, but the demonstration effect after the data was integrated was incorrect