SlimeVRX/SlimeVRX #57: Hello.SlimeVRX,I'm very confused to reproduce PIP

yuhuison, 2023-08-12:
Hello，SlimeVRX , I am interesting in mocap project of VR such like SlimeVR.I found so are you very active in it.

I haved diy some piece imu using the chip bno085. But when I want to use it to reproduce PIP, I found the result is not very good.

Can u help me? I think we can share ideas and fruits and I can post some imus to you for testing.We can use Wechat to communicate.

SlimeVRX, 2023-08-12:
Please add me!

[image]

SlimeVRX, 2023-08-13:
I realy need BNO085 sensors for testing

yuhuison, 2023-08-13:
Hi SlimeVRX, I have send  a friend request to you.

closed 2023-08-18 by yuhuison

SlimeVRX, 2023-09-13:
Hi @yuhuison, I lost my old Wechat account, can you help me authenticate it? Please send number 766253 to my account

SlimeVRX, 2023-09-13:
Sorry for the inconvenience!

yuhuison, 2023-09-15:
i have send it !

SlimeVRX, 2023-09-15:
Many Thanks!

SlimeVRX, 2023-09-26:
Hi @yuhuison,

I failed to get my old Wechat account back.

Are you planning to continue supporting the BNO085 sensor for my team, we are short of sensors for testing. We really need your group's sponsorship.

Many Thanks!

Tin Nguyen

SlimeVRX, 2023-10-03:
Hi @yuhuison,

We have the same goal, we promise to support you as much as possible!

yuhuison, 2023-10-04:
oh,sorry , I am now so busy for my school because new terms begin，In fact ， our hardware is not stable and we are fixing some problems , so it is not can be use now, but I'am glad to share the progress of it, and I promise that I will fulfill it to help yours. how can i contact of you ？ do u have another wechat account or any other IM app?

yuhuison, 2023-10-05:
@SlimeVRX

SlimeVRX, 2023-10-05:
Hi @yuhuison 

Sorry, My current job is also very busy, when I have free time I will work on mocap.
My Wechat is broken, I can't get back my old account with my phone number. 
Do you use Skype, Slack, Twitter or Discord?

SlimeVRX, 2023-10-07:
Hi @yuhuison

I see you mentioned that the hardware is having problems, I can help you with hardware programming. I think I can assist you a lot but I also don't have much time. So if possible, let's make a plan together and execute it.

SlimeVRX, 2023-10-12:
Hi @yuhuison. Please check help me!

SlimeVRX, 2023-12-10:
Hi @yuhuison. I need sensors for experiments, Can you sell me 6 BNO085 sensors?

SlimeVRX, 2023-12-11:
Hi @yuhuison. I really sensor for testing, Please sell me 6 BNO085 sensors?

SlimeVRX, 2023-12-11:
Hi @yuhuison. 

I really sensor for testing. Our sensors are currently inaccurate. 
Please sell me 6 BNO085 sensors?

Thank you very much!

Best wishes

Tin Nguyen Thanh